LumaTeam/Luma3DS #308: Luma3ds and payloads won't load when switching from 32 to 128gb sd card

ismael587, 2016-12-31:
Hello!
I first installed luma 3ds 6.5 cfw on my 32gb sd card and everything works fine. I bought a 128 gb sd card and i formatted it to fat32. I copied every files on the 32 gb card to the new 128gb sd. But now i can't load luma3ds from the sd and i can't get payloads working (hourglass9). However luma3ds still load from the nand when i hold select at startup. It seems that luma 3ds doesn't recognize my 128 gb sd card but everything works fine works fine in the 3ds menu.
I don't know what to do?

urherenow, 2016-12-31:
Were you using emunand before? Simply copying the files doesn't copy emunand/rednand.

If no, try switching your format between 32k and 64k clusters.

ismael587, 2016-12-31:
i was not using emunand
i tried switching to 64kb clusters but it stil does not work

ismael587, 2016-12-31:
There's no problem for installing cias with FBI the card is recognized

ismael587, 2017-01-01:
finally i reformatted with 32kb clusters with another tool and i made a direct copy from one card to the ather and it works fine now.

james-d-elliott, 2017-02-08:
You should be able to close this then.

weirdal3333, 2017-02-09:
This problem usually happens when the card's partition table comes with a GPT partition table. All you need to do is change it to a MBR partition table and format it as a FAT32 partition, and it should work fine.

closed 2017-02-09 by TuxSH