LumaTeam/Luma3DS #310: Reboot patches broken on Safehax

idgrepthat, 2017-01-04:
Extra memory games will not work when launching luma3ds as a payload with Safehax.

Margen67, 2017-01-04:
Why wouldn't you just install A9LH?

idgrepthat, 2017-01-04:
> Why wouldn't you just install A9LH?
Because I'm a safehax peasant ;p

Really, I'm asking on the behalf of others who are scared of attempting any risk of bricking.

Margen67, 2017-01-04:
Tell them to stop being paranoid.
Staying on 11.2 puts you at risk of updating SysNAND without any form of protection.

Mrrraou, 2017-01-04:
"lol why firmlaunch no work when i use firmlaunchhax" blah blah blah
you need to change the payload's path and stuff. like you'd do on a k9lh setup.

idgrepthat, 2017-01-04:
> "lol why firmlaunch no work when i use firmlaunchhax" blah blah blah
you need to change the payload's path and stuff. like you'd do on a k9lh setup.

Ok, I've been dressed down by the Black Cat. That's the fat lady singing for my issue. Closed. :-)

closed 2017-01-04 by idgrepthat