LumaTeam/Luma3DS #311: Failed to mount CTRNAND

MrWiik, 2017-01-04:
Well im trying to get a9hl and luma 3ds on 11.2.35E with safehax and when i start the 3ds i get failed to mount ctrnand

admiralkikan, 2017-01-05:
If you were using Phailects guide during the process it tells you to proceed to the next step if you get that error.

travjmaggs, 2017-01-06:
What about after everything? I have everything done but need to go back into Luma settings to edit something, will I get the failed to mount CTRLNAND again? Every time? If so, THEN how would i get out of it? Or should it not say the error after? Any advice please!

admiralkikan, 2017-01-06:
You shouldn't be getting that error at all except that one time. I have no idea what you did.  You could always start over.

travjmaggs, 2017-01-06:
Thank you, I was just scared to go into those settings again, but it worked fine.

admiralkikan, 2017-01-07:
No more issues? Ctrnand fail to install doesn't pop up anymore?

travjmaggs, 2017-01-07:
Nope have used it twice with no problem, thanks.

closed 2017-02-12 by TuxSH