LumaTeam/Luma3DS #312: Config non-save

idgrepthat, 2017-01-05:
When I change something in the config menu, hit start (save), then immediately go to the payload menu, whatever I changed is not saved after boot to home menu.

urherenow, 2017-01-05:
Did you save settings to nand at some point and now saving with SD inserted?

idgrepthat, 2017-01-05:
Never.

TuxSH, 2017-01-10:
Can't reproduce

Mrrraou, 2017-01-10:
had the issue when chainloading to luma3ds-rosalina, so doesn't seem to be related to different luma versions it seems

idgrepthat, 2017-01-10:
@TuxSH Bug is still present on latest commit.

Exactly how to trigger: Boot 3ds while holding select, change something in config menu, tap Start briefly, then immediately hold Start. This will bring up the payload menu of course. Do whatever you want then come back to the Config menu and nothing will be changed.

IcySon55, 2017-01-22:
I can confirm that this is the case. I ran into it a few times while installing A9LH on a few systems.
It's exactly as @idgrepthat described.

I install A9LH then shutdown, boot into Luma settings, tap start, then quickly hold start to launch Hourglass and restore the sysnand backup. Reboot, and I'm greeted with the Luma settings again.

AuroraWright, 2017-04-14:
Fixed

closed 2017-04-14 by AuroraWright