LumaTeam/Luma3DS #313: region emulation

admiralkikan, 2017-01-05:
As it appears, region emulation for whatever reason still seems to be broken whether I physically make the text in the right location or use the revised locale switcher. I know that this issue has been brought up before just thought I'd bring it back up. Issue is in both nightly and stable 6.6.

Sn0wCrack, 2017-01-08:
Have you checked you've enabled locale emulation your Luma settings? This ended up being my problem.

admiralkikan, 2017-01-09:
Of course, that's why i brought up the issue for both releases. I haven't changed any settings or anything when i had originally updated to 6.6 but for whatever reason now on my 2nd n3ds 6.6 doesnt emulate and on my 1st n3ds nightly doesn't either. Im just wondering why it's broken all of a sudden.

AdowTatep, 2017-01-14:
I'm having the same problem. I used the Luma locale switcher cia and when opening the generated files from pc everything seems all right, I did enabled the emulation on settings but it is not working in games that I remember working before

admiralkikan, 2017-01-14:
Adow, I've come to realize that the r

admiralkikan, 2017-01-14:
I've come to realize that the rrgion emulation does indeed work. However, the language emulation portion does not appear to work as it should. I would know since i use pkmn to test this stuff out.
Roll back to 6.5 if you want both to work as they should.

AdowTatep, 2017-01-15:
@admiralkikan i can't download the 6.5 version, and i don't think it's in https://astronautlevel2.github.io/Luma3DS/

Margen67, 2017-03-06:
https://github.com/ericjwg/LumaLocaleSwitcher/releases

closed 2017-04-17 by AuroraWright