LumaTeam/Luma3DS #315: Random Crashes w Dump Files Being Created

smelly33, 2017-01-06:
Hello,
Running N3DSXL 11.2.0-35U (LATEST LUMA + A9LH)
The issue arises when playing CIA games or when entering or exiting out of default apps on the device
The device would be fine for 5 - 20 minutes before freezing and turning black. I have run h2test to see if the SD card is the issue, and the SD card comes out fine. I have done a nand restore and updated back, the issue is still there. The system sometimes (rarely) created dump files which I attached below (github did not allow me to upload zip files) 
https://ufile.io/41ce3
https://ufile.io/7b6b4

smelly33, 2017-01-10:
https://transfer.sh/DpPAl/crash-dump-00000004.dmp

It created another dump when trying to boot (I am not using any Dev features)
It would be great if someone can help me read what these dump files say

TuxSH, 2017-01-10:
Could maybe a file system issue; could you try to, at first, dump your CTRNAND partition using decrypt9?

smelly33, 2017-01-10:
Thank you for the help, and yes it created a CTRNAND.bin file.

closed 2017-04-27 by AuroraWright