LumaTeam/Luma3DS #318: top Splash screen not loading

fennectech, 2017-01-08:
the file /luma/splash.bin is not being put on the top screen   only the bottom splash works    top is simply blank

RefusaI, 2017-01-08:
Are you sure the dimensions are correct? For the top screen, it should be 400px horizontal and 240px vertical.

fennectech, 2017-01-08:
How do i check the dimensions of a .bin file?  i downloaded precreated one

fennectech, 2017-01-08:
ill upload my bin file 
[splash.zip](https://github.com/AuroraWright/Luma3DS/files/692514/splash.zip)
  This is the mac os7 theme  i want to use   the bottom one works  but top does not

fennectech, 2017-01-08:
if you could return it to a .png i can fix it myself   but im unsure how :/

fennectech, 2017-01-09:
Well i cropped the preview and got it to work  :D   will close this

closed 2017-01-09 by fennectech