LumaTeam/Luma3DS #323: [N3DS] "Clock+L2" shows black screen for a Shin Megami Tensei game, "Clock" does not

maki-chan, 2017-01-11:
So, there is my first problem with the Clock+L2 setting on my N3DS.

When I try to start Shin Megami Tensei: Devil Survivor Overclocked EUR (Title ID 000400000009c000), the ATLUS publisher splash screen shows and then it turns to a black screen while playing the music and sound of the intro cutscene, but it does not show the cutscene itself and the game is not reacting to any button input (while you can definitely skip it with a button when the intro shows normally).

When I start Luma3DS with the setting "Clock" in the configuration, the intro shows normally and the cutscene can be skipped.

An idea to fix this for multiple games instead only this one would be to implement the possibility to set the CPU clock switch on a per-title basis if that's possible at all. Another idea would be to hold a button of my choice on startup to load an alternate config file (which in my case would have other clock settings).

It's inconvenient to switch that setting whenever I start that specific title and afterwards when I am done playing it.

dogtopus, 2017-01-13:
This game is always problematic on New 3DS mode. It has nothing to do with luma

Margen67, 2017-02-12:
#357

TuxSH, 2017-02-14:
> Per-game New 3DS mode configuration sounds good, though.

Not possible right now

closed 2017-02-14 by TuxSH