LumaTeam/Luma3DS #324: Failed to mount CTRNAND; Can't access hourglass9/decrytp9WIP only shutdown

huygreat1996, 2017-01-12:
Just like the title, i cant do anything except access luma setting, press start and it happen, try to restore sysNAND through hourglass but i neither can access it nor decrypt9WIP. Help pls!!!
P/s: i've already tried to re-make luma folder and run hourglass9 as armloaderhax... nothing happen. I think for some reason it cant boot the payloads' files

weirdal3333, 2017-01-12:
So, your directory structure looks like this

/
  luma/
             payloads/
                               x_hourglass9.bin
                               y_decrypt9.bin
  arm9loaderhax.bin
  ..... Everything else 

And you only hold x or y to boot into hourglass9 or decrypt9, respectively.
Also, check out the closed issues.

weirdal3333, 2017-01-12:
My bad, I thought the spaces would show up

/payloads/x_hourglass9.bin
/payloads/y_decrypt9.bin
/arm9loaderhax.bin

huygreat1996, 2017-01-12:
Yeah the structure is like that. What do you mean when you said "you only hold" there is another method? Because i tried by pressing X or select or up + power (i named those buttons)

weirdal3333, 2017-01-12:
maybe download new payloads? Also, I forgot to make it /luma/payloads/x_hou... Etc.

piratesephiroth, 2017-01-12:
should it be luma/payloads?

weirdal3333, 2017-01-12:
Yes, exactly. Also, you should be following the guide at http://3ds.guide/ and asking on gbatemp about your problem...

huygreat1996, 2017-01-12:
@weirdal3333  Already tried changing payloads yet no help.
@piratesphiroth It wont matter friend, i tried put the payloads folder on root and in luma folder, my n3ds just cant boot those .bin files

huygreat1996, 2017-01-12:
@weirdal3333 Already done... there are some guys have this problem too
https://gbatemp.net/threads/a9lh-installation-failed-to-mount-ctrnand.452037/page-3

weirdal3333, 2017-01-12:
You have to configure Luma and press start first. Then you can use the payloads.

weirdal3333, 2017-01-12:
Also I read that thread, and if you can boot into Luma (and have arm9loaderhax already) you can use safemode and do it over again.

huygreat1996, 2017-01-12:
@weirdal3333 Done mate, of course i config it then i load the payloads... Config => start => Failed to mount CRTAND / press any button to shutdown => start + power => press any button to shutdown / x +power => same thing. Recovery mode cant access: A + L + R + Up + power and it just show that screen again

maki-chan, 2017-01-13:
Failed to mount CTRNAND happened on my New 3DS because Luma3DS cannot use the system at version 2.1.0 after ctrtransfer. You have to restore your NANDmin.bin backup or use the 9.2.0 ctrtransfer files to restore your system to a newer version. All of that is mentioned in the guide at https://3ds.guide, though.

MelonGx, 2017-01-21:
Why asks question but not reporting Luma issues here?
This is not a place for FAQ.

closed 2017-01-25 by TuxSH