LumaTeam/Luma3DS #325: copying content of original (16GB) card to upgraded (32 or 128 Gb) card black screen

pcs3rd, 2017-01-12:


pcs3rd, 2017-01-12:
luma3ds + a9lh (sysnand)
because of current personal issues, I am not able to keep at least 4 GB of space available on my 16GB card, so, I tried copying it to both a 32GB card and a 128GB card, both I tried formatting with godmode9 in both 32k and 64k clusters, and eventually "zeroed the sdcard" using windows and also linux. no matter what, I have not gotten my O3ds to boot from any sdcard other than the original 16GB (anything else, the blue power light just stays on, but there is no sign that the ds is actually booting from a9lh.bin (the screen lcd stays off, and I cannot boot into any of the payloads), and I am just copying the files from the working card to a larger FAT32 ones, but again, all I get is a black screen, not even the wireless status light on the 3ds slider light.)

urherenow, 2017-01-12:
This isn't a help forum. Ask on irc or gbatemp.

weirdal3333, 2017-01-12:
It might be because you need the 'arm9loaderhax.bin' and not a 'a9lh.bin'
Do yourself a favor and follow the tutorial on http://3ds.guide/ again.

Basically, you copy the 'arm9loaderhax.bin' (the one that works) to the CTR NAND and it will boot without an SD card.

weirdal3333, 2017-01-12:
Yea... I think the guide should be pinned at the top or something, so that there are less 'noob panics'

pcs3rd, 2017-01-12:
`I aready went to gbatemp, for a entire 2 days, everything recomended did NOT work. I am considering this a problem because everyone claims trhat I just format fat32 and copy everythging over, I have checked cluster size, format, partition markings, primary partitions, and I wouldn't be here if gbatemp users help solve the problem.

weirdal3333, 2017-01-12:
arm9loaderhax.bin has to be from the releases. Maybe it got corrupted while copying? Copy the new one and hold select while booting.

pcs3rd, 2017-01-12:
could I have a direct download please? using chromium on a raspi is a pain in the a55

weirdal3333, 2017-01-12:
It has to be a MS-DOS partition table btw, not gpt...

Suggestion to AuroraWright: implement gpt partition table support for the 3DS.
Imagine.... A 2 terrabyte SD card!

pcs3rd, 2017-01-12:
it is.

pcs3rd, 2017-01-12:
i am in godmode9 @tm, so I can jkust copy the a9lh bin file to ctr nand?(arm9loaderhax.bin)

weirdal3333, 2017-01-12:
Umm... https://github.com/AuroraWright/Luma3DS/releases/ 
You can find a release on your phone and copy the URL/link address of the download...

Why isn't this common knowledge?!?!?!

pcs3rd, 2017-01-12:
but I only have access to a raspi right now, I do not have a non-brewmp phone

weirdal3333, 2017-01-12:
Come back later after you've formatted the new SD card with a fat32 msdos type partition table and finished with the guide.

pcs3rd, 2017-01-12:
*I'min western PA*... it is going on 10. I am not exactly able to just boot up a windows laptop atm

pcs3rd, 2017-01-12:
but again, I am confirming that I can just copy my a9lh.bin file to the root of my ctrl nand

weirdal3333, 2017-01-12:
Nope, it's the arm9loaderhax.bin file from the Luma 7z file.
 But you should follow the guide first, as it explains why you are doing this.

pcs3rd, 2017-01-12:
both a9lh files are the same size, I have tried this on 2 different machines on 2 different windows boxes an d2 diferent linux boxes. It isn't just a coincidence that the only card that continues to work no matter the enviroment is the original. when I say a9lh.bin, I am referring to arm9loaderhax.bin!

pcs3rd, 2017-01-12:
look. I understand how easy it is to make a electronic (cup) coaster/paperwight  from playing with tate. the last thing I need to be doing is explaining why I have bricked a 3ds.

pcs3rd, 2017-01-12:
*seems to be working booting ctrnand. I will try to update the luma3ds cfw on the 32gb card will be back if need further help, the guys over @ gbatemp haven't helped me as much as I, the lost little idiotic noob am. Thank you? ( for now)

pcs3rd, 2017-01-12:
i am also assuming I can safely run luma3ds updater safley, correct

weirdal3333, 2017-01-12:
An excerpt from the http://3ds.guide/ front page

> Your SD card should be MBR, not GPT (the SD card that comes with the device will be MBR by default).
If you need to format a brand new SD card, you can use guiformat and set to an Allocation Unit Size of 32K.

So, don't forget about changing the PARTITION TABLE to MBR (or msdos if using gparted). Then format the partition in the MBR partition table as Fat32 with an allocation unit size of 32K.

weirdal3333, 2017-01-12:
I think that it only works when you have a functional SD card. If the SD card works in the 3ds, then go ahead! The CTR NAND arm9loaderhax is for emergencies only, so fix the SD card. Also, what system menu version are you?

urherenow, 2017-01-12:
@weirdal3333 um... Fat32 actually can handle 2TB as it is. That's the maximum though.

weirdal3333, 2017-01-12:
Oh, well... How big can the largest gpt partition hold? Imagine that instead!

Also if gpt partitions were supported... We would have less of these threads.

tlhv, 2017-01-20:
So I'm new to all the 3ds CFW and I installed everything step by step off that guide. I have everything working perfect but now I want to upgrade my 16gb card to a 64gb card. Can I just format my 64gb to Fat32 and transfer everything over from the original 16gb to the new 64gb or is there a specific back up and restore process I have to follow?

Shadow322, 2017-01-20:
This isn't a help forum close this and go ask on gbatemp

Margen67, 2017-01-20:
@Shadow322 This isn't tlhv's issue to close. @pcs3rd or a dev needs to close it.

MelonGx, 2017-01-21:
Too @pcs3rd and all noobs:
1) RTFW = Read the fucking wiki
2) This board is created for reporting Luma issues, but not asking questions.
If you want to ask question, GTTFGT = Go to the fucking GBATEMP thread.

closed 2017-01-26 by TuxSH