LumaTeam/Luma3DS #328: RomFS redirection not working for cartridge games

will0n, 2017-01-13:
System got stuck in "Nintendo 3DS" loading screen every time. The cartridge game I've tried are Fire Emblem: Fates and Persona Q.
I can confirm that HANS works properly with the same romfs files. I've tried some of my digital download games from eShop and the redirection work properly as well.
I'm using the nightly build from http://astronautlevel2.github.io/Luma3DS/builds/Luma-1eb18c17.zip

configchan, 2017-01-18:
I'm having a similar issue where it crashes attempting to load a redirected romfs. If it helps, here's the crash dump I got from enabling errdisp.
[crash_dump_00000003.zip](https://github.com/AuroraWright/Luma3DS/files/713539/crash_dump_00000003.zip)

will0n, 2017-01-19:
@configchan oh you reminde me of dump the game to CIA. CIA doesn't work either. BTW, how to get crash dumps?

configchan, 2017-01-19:
Enable errdisp in luma config.

closed 2017-02-09 by TuxSH