LumaTeam/Luma3DS #331: Luma3dsv6.6.7z link not working

Darkhowl, 2017-01-14:
The Luma3dsv6.6.7z isn't working for me, it takes me to a page that says "This XML file does not appear to have any style information associated with it. The document tree is shown below." Here's a picture of the issue. - http://imgur.com/a/sSFEw

Darkhowl, 2017-01-14:
Really need this link to fix my 3ds with arm9loaderhax.bin, please fix the link asap!

Shadow322, 2017-01-14:
https://astronautlevel2.github.io/Luma3DS/builds/Luma-1eb18c17.zip

zhi1ong, 2017-01-14:
https://www.mediafire.com/folder/11qzu54cam51w/Luma

Size = 217kb, same as author's release file.

TuxSH, 2017-01-14:
Should be fixed.

The problem is due to what happened last Thursday.

closed 2017-01-14 by TuxSH

Darkhowl, 2017-01-14:
The link works fine now, thanks everyone.

AdowTatep, 2017-01-15:
past versions aren't working, want to get 6.5 because 6.6 language emulation is not working

Shadow322, 2017-01-15:
Does it work in latest nightly?

https://astronautlevel2.github.io/Luma3DS/builds/Luma-1eb18c17.zip

AdowTatep, 2017-01-15:
@Shadow322 the language emulation? it still didn't using jap system with us bravely default using the latest nightly, the game is still in japanese because of the system language, even though it's an us game

TuxSH, 2017-01-15:
Which game?

AdowTatep, 2017-01-15:
I just said .-. .  American bravely default with language and region set manually to US still show japanese language in japanese 3ds because it has those files. I remember it working in past versions of luma but I updated to play MH online, fix that friend thing and then the language emulation stopped working.