LumaTeam/Luma3DS #332: romFS .ips possible?

Onoitsu2, 2017-01-15:
I know that we have code.bin IPS files possible, however was wondering if romFS IPS files were possible to be used. For example something like packing http://gbatemp.net/threads/maplestory-3ds-the-girl-of-destiny-100-eng-patch-release.430440/ As being able to use an IPS would be wonderful, so I don't have to mod the romFS, a 400+ MB file, if a 63.1kb (in current testing) file could be used instead.

dogtopus, 2017-01-15:
IPS doesn't support files larger than 16MB. End of the story.

Mrrraou, 2017-01-16:
@dogtopus you should learn to read

Marma92, 2017-01-16:
blackcated.

Onoitsu2, 2017-01-16:
@dogtopus I don't see how IPS supports only 16MB files max, because certain utilities may have that limitation but IPS XP does not, it gladly accepted my over 400MB (approx 424MB) file, and provided a patch bearing a size of 63.1KB, after several minutes processing time. I could understand the Luma itself having a limitation of altering up to a 16MB file, due to memory constraints of the O3DS devices for example. But it is definitely possible for an IPS to be used on larger than a 16GB file, it is simply a patch and technically could be used on any size file, it is only a listing of address to be modified and a chain of bytes that are to be modified to.

dogtopus, 2017-01-16:
@Onoitsu2 If I recall correctly, one of the reason UPS was implemented was IPS only supports input files up to 16MB due to its 24bit file offset. So IMO if a tool supports a IPS patch on input files larger than 16MB, it is either somehow splitted the input file into section, or it uses non-standard IPS format.

dark-samus, 2017-01-22:
@dogtopus it can handle files bigger than 16 MB, however only bytes in the 16MB range can be affected by the IPS patch. So if I had a 1 TB file, and only changed things within the first 16MB, I could still patch said 1 TB file

vaguerant, 2017-02-12:
@Onoitsu2 As someone who's worked on the MapleStory hack, I can assure you that there are more than 63KB of changes. I suspect it's as dark-samus describes; there may be 63KB of changes within the first 16MB. But then there's the other several hundred megabytes that are changed that are just not being included in your patch. This whole issue is predicated on misinformation.

closed 2017-03-05 by TuxSH