LumaTeam/Luma3DS #334: 3ds 4.5 emunand freeze

Makein, 2017-01-17:
I want to try Luma3ds before installing it on sysnand.
I copy this files to root of SD cart:
arm9loaderhax.bin
Luma3DS.dat
luma\firmware.bin
luma\cetk

Place MSET on NDS cart
Then boot mset in nds and choose 4.x
Then go to settings->profile->DS profile
And freeze.

Is it right?
Because for me it don't work =\
Oh, forgot to mention: i have RXTools emunand on last frimwhare

Margen67, 2017-01-17:
http://3ds.guide/

Makein, 2017-01-17:
Like i say, i dont want install it on SysNAND for now.

urherenow, 2017-01-17:
Github isn't a forum. Go somewhere else.

Makein, 2017-01-17:
Well, in this: https://github.com/AuroraWright/Luma3DS/wiki/Installation-and-upgrade
It say i can use it on 4.3 and problem go from there. 
I don't see why it not issue, because it's not working or have bug

Margen67, 2017-01-17:
>Like i say, i dont want install it on SysNAND for now.

Why not?

Makein, 2017-01-17:
1. Because I need update 3ds and then downgrade it.
2. I would like to use many ecfw for now
3. Choose what fw is better
4. Future gw without cart cheat menu hack... + ntr
5. Etc

Margen67, 2017-01-17:
1. So? If it wasn't safe it wouldn't be in the guide.

2. rxtools is obsolete, outdated, and inferior to Luma and every other CFW
Luma is the best CFW in most cases.
rxtools however is objectively worse.
It's tools are buggy, and broken. they've been superseded by tools like Decrypt9WIP, etc.

3. The best fw is A9LH updated SysNAND.

4. You can still use EmuNAND with A9LH, but there's not much need to.
Are you implying emuNAND is more safe than A9LH sysNAND? Because it's not.

5. What?
Gateway will always require the cart.
NTR works fine with Luma.
NTR also has cheats.

Makein, 2017-01-17:
1. Is it? Are you sure? Even updating ofw is not safe, because who know how console can glitch
2. I know it about Rxtools, but it still works great and run all games. If you read my post fully, then you saw that I say about using many eCFW
3. I dont need SysNand cfw, because who knows when i wanna sell it or **** (have some other reasons that would go with Nintendo license)
4. I know it too.
5. I know it too. But, GW functions can be used with other cheaper cart with plugin (if it's like that, then why somebody cant port them?)
6. 
If we gonna talk, then it's become forum...
I ask for solution and many come to talk...
Ugh.

Margen67, 2017-01-17:
1. Yes. Nobody would make a guide for something that bricks your 3DS. The only real danger from updating sysNAND is losing access to exploits/hacks.
Most if not all A9LH bricks are from user error.

2. eCFW? Do you mean CFW? The only relevant CFWs I can think of are Corbeniks, and Luma. I'm not even sure if Corbeniks supports launching from 4.x since it's so old.
Pasta, Palantine, etc. are all almost completely pointless to use nowadays.
You should just switch to Luma+A9LH. It's not even hard, and is your best choice.

With MSET you have to go to settings etc. which is quite inconvenient in comparison to A9LH which has the system coldboot with CFW.

3. You can uninstall A9LH. It's objectively better than emuNAND.
You can still sell systems with CFW.

5. Gateway patched it. Also flashcarts are a waste of money nowadays. Don't buy them.
Somebody could potentially make Gateway work without the cart (or bricking) but it's very unlikely and there's no point since Luma is better anyway.
Gateway cheats also wouldn't work without the cart without further patches since cheat results are saved to it.

6. What do you expect from such an ancient/outdated setup like that?
Either you:
A. Get your old, outdated, inconvenient, and unsafe method of booting CFW working again
B. Switch to Luma+A9LH which is better in every way, and future proofs your system.

How is it unsafe?
You start up without any protection whatsoever. If you go to system settings and update your sysNAND you're screwed.
Also if you brick your sysNAND you won't be able to recover from it.
Also rxtools' tools are buggy, and broken.
Unlike A9LH which lets you use updated sysNAND and lets you unbrick without a hardmod.

Seems like a pretty easy decision to me.

Makein, 2017-01-17:
Does i need repeat myself?
Because of some legit reasons i CANT do it.
eCFW - Emunand for me, because it's emulated cfw
And like i say it upper, because of it i can use ONLY emunand...
If it's impossible to boot luma3ds then say so already.
Because i don't see any reason to read the same answers again, but i'm glad that you try persuade me to use it.
PS: GW not fully patched it. It's  like nintendo release update to patch hack, but with some manipulations it can work again

You can still sell systems with CFW. - Where I leave, for this, you can be in jail for 5-10 years. Nice, right?

Margen67, 2017-01-17:
Anyone can do the guide. Any system can be hacked now.
You don't absolutely need to use emuNAND. The guide doesn't even need them.
You don't need to update before starting the guide.
You won't lose anything unless you do something stupid.

Also most if not all bricks are user error.
If you think otherwise do some research, then come back.
You will find that most bricks are due to user error not random bricks from the software.

Once you've done the guide you can do whatever you want including restoring the emuNAND if you want to.
https://3ds.guide/
Updating on A9LH is safe since the exploit is installed to NAND and is write protected so it won't be lost after an update.

Booting Luma on 4.5 is irrelevant since the guide doesn't require it at the start.
You install Luma near the very end.
If the emuNAND works in rxtools it will work in Luma.
Are you following a different guide? Because you shouldn't.

Then like I said before uninstall A9LH before selling the system.
I honestly don't know what you're talking about near the end.

gnmmarechal, 2017-01-18:
Huh, it's NOT eCFW, that's a term used in the PS Vita community to refer to a CFW for the ePSP (aka the PSP emulator in the PSV). Also, EmuNAND is not an emulated NAND either. It's a redirected NAND as far as I'm aware.

closed 2017-01-18 by TuxSH

Margen67, 2017-01-18:
https://www.reddit.com/r/3dshacks/comments/5op3st/how_these_bricks_happen/
https://i.redd.it/elzf71hutgay.png