LumaTeam/Luma3DS #335: Region/Language emulation and Code.bin patches broken with the latest commits

Shadowxp13, 2017-01-19:
Hi, after using the hourly builds, me and four other people noticed that issue.
It has something to do with the last commits, because it works with the latest stable version (6.6)
Luma simply ignores it, acting like there was not any file in the locales folder and/or code_sections folder.
Some people also reported on the luma thread in GBATemp, but i think that posting here is better.

FateForWindows, 2017-01-19:
I'm experiencing the same issue as well. I noticed it when trying to play Smash Neos with Saltysd.

DeckOfNoobs, 2017-01-19:
Luma3DS nightly changed the Code Sections folder to titles/(titleidofgame)/code.bin

Shadowxp13, 2017-01-19:
What about region emulation?
It is (or at least was) in luma/locales/TITLEID.txt

Shadowxp13, 2017-01-19:
Got it, (luma/titles/titleid/locale.txt). anyway this should be added to the wiki

ericjwg, 2017-01-19:
https://github.com/ericjwg/LumaLocaleSwitcher/releases

support 6.6 nightly

closed 2017-01-25 by Shadowxp13