LumaTeam/Luma3DS #336: Swapdoodle not opening

renatosk, 2017-01-19:
My luma is in the latest version , tried install from eshop , from a cia and still the game doesn't open , he stuck in that nintendo 3ds screen , do you know any fix? (sorry for the bad english btw)

Margen67, 2017-01-19:
What entrypoint are you using?
Are you using emuNAND?

OneCrazyRussian, 2017-01-19:
Even if it's not stated in eshop or elswhere, swapdoodle won't work on older 3DS firmwares

Was fixed for me by updating from 10.3 to latest FW on Luma+AL9H 

The update is included in installation guides almost everywhere but you could've missed it or get lost in multiple different emuNANDs

Margen67, 2017-01-19:
If you're still using emuNAND you should switch to updated sysNAND http://3ds.guide/

renatosk, 2017-01-19:
Fixed by deleting the firmware.bin from the luma folder , anyway thanks guys

closed 2017-04-17 by AuroraWright