LumaTeam/Luma3DS #340: [Suggestion] An in-game menu.

Monotek18, 2017-01-20:
Since a lot of time, GW have his own in-game menu.

So I am wondering if Luma would have that one day, but without having to boot ntr...

noX1609, 2017-01-20:
Nice idea 👍 
And if they could implement a cheats engine that can use Gateshark Codes but i think that is a dream that never will become true :( If Gateway were open Source than maybe someone would implement it but because it's not you would have to reverse engineering the Gateway Card / software at first.

Mrrraou, 2017-01-20:
i doubt there'd be any kind of cheating if that were to be done and put inside luma

Margen67, 2017-01-20:
You can convert GW cheats for use in NTR

Monotek18, 2017-01-21:
luma/cheats/**TitleID**.txt/plg ?

And the comment of @Margen67 is right, Gateway and NTR have their + and -.

Like Gateway is stable, we don't need to boot an app on the home menu when we want to use it.
But closed source (of course) and not having regular update.

NTR can be better since it load game plugin written in C, so more possibilty than the AR codes type.
Infinite loading screen when we change to an other game.
Not open source, and update are stopped.

That why we need an open source replacement.

Mrrraou, 2017-01-21:
why the fuck would you reverse ntr lol

Monotek18, 2017-01-21:
because he have an in-game menu, not as good as Gateway, but he have what we want to be open sourced.
(or correct me if I am wrong)

Nanquitas, 2017-01-21:
The menu doesn't really need NTR.

Margen67, 2017-01-21:
It would be nice to have an open source NTR replacement regardless

Mrrraou, 2017-01-21:
@Nanquitas it indeed does not at all

AnalogMan151, 2017-01-23:
@Nanquitas Thanks for that confirmation. All we really need is a feature that could load a plugin file. Could go in SD:/luma/titles/<gameID>/*.plg

If it was part of loader, then it would even work on o3DS in extended memory mode if I'm correct in my understanding. A new plugin framework would be needed that doesn't rely on NTR, but that's easier than reverse engineering and making a new NTR from scratch. 

While not a necessary feature request, it would definitely be a well received one that would make Luma3DS even more popular than it already is.

Monotek18, 2017-01-23:
@AnalogMan151 That could be a good start, I don't think it's as hard as making an in-game menu, but a good start.

adrifcastr, 2017-04-05:
I wish someone would try looking into this

movr1r1, 2017-04-24:
I don't want to be that guy, but I am not sure this will happens.
Not because "cheating is bad", but because of how hard it is.
Gateway did that because $$$, but dev that do that for free won't have the same motivation.
Luma have classic feature like patch and ect, but an in-game menu is very a advanced stuff and you'll need to be good at RE.
I wish Luma would have that too.

Dimensional, 2017-04-24:
If Luma was to add in an open source cheat menu, it would cut into Gateways primary feature, and give people even more reason to not buy from them. Right now the in game menu is the primary reason people still use Gateway, despite so many problems like no Firm0/1 protection.

From what I can figure, it appears the in game menu feature relies on various hooks with App IDs in the system, similar to Ntr. Whatever Gateway does, it appears they patched a file or injected some code into a system process to allow for their own hooks, mimicking the pause feature of the home button while in fact calling another menu all together.

movr1r1, 2017-04-25:
of course they patched/injected stuff, otherwise no way to have a such thing working

AuroraWright, 2017-04-25:
Gateway has an arm9 thread running at all time, that's also how they do the game selection

jaeman109, 2017-06-05:
lol now it's here in luma 8

TuxSH, 2017-06-05:
(∩ ͡° ͜ʖ ͡°)⊃━☆ﾟ

closed 2017-06-05 by TuxSH

Monotek18, 2017-06-05:
My prayor has been heard !

That cool and all, but cheat search menu/plugin loading when (∩ ͡° ͜ʖ ͡°)⊃━?

jonirrings, 2017-06-05:
sorry to bother, but what does this 颜文字 (∩ ͡° ͜ʖ ͡°)⊃━☆ﾟ resemble?
Rosalina & luma (ロゼッタ&チコ, Rosetta & Chiko)?

Monotek18, 2017-06-05:
@jonirrings It's "magic lenny"
A meme ;)