LumaTeam/Luma3DS #344: Enable game patching not working

Mephes, 2017-01-22:
The Enable game patching option won't load code binaries anymore. Everything worked fine on the last version of Luma, so why isn't it working here?

sora10pls, 2017-01-22:
make sure your directory is updated

example
`luma/titles/<title_id>/code.bin`

thunderkai, 2017-01-29:
I am having this issue with Smash 3DS too....
Mods are not loading or the language emulation isn't working either

sora10pls, 2017-01-29:
Smash 4 mods working just fine for me

language emulation example (Smash 3DS USA)
`luma/titles/00040000000EDF00/locale.txt`

sora10pls, 2017-01-31:
I think this issue should be closed. We've basically just confirmed that the feature IS working as intended, people just need to update their directories.

closed 2017-02-09 by TuxSH

dothackjhe, 2017-06-22:
How does one get the "code.bin" for each game needed for game patching?

WubTheGame, 2017-06-22:
@dothackjhe 1: this isn't a support forum, even i'm still learning that. 2: try the Smash 3DS Hacking Discord

dothackjhe, 2017-06-23:
@WubTheGame Never mind the question. I just learned that what I was asking for was the old method of making game patching work with Luma3DS. As with the latest release of Luma3DS (B9S), the feature runs with "boot.firm" from the root of the SD/micro-SD Card.