LumaTeam/Luma3DS #345: Issues Booting Up System With Latest Luma 3DS Commit

Pepy550, 2017-01-22:
Having some trouble powering on my system. For some reason, whether it's rebooting from Decrypt9 payload or just powering on, I get a black screen with only the blue LED.

Margen67, 2017-01-22:
https://github.com/AuroraWright/Luma3DS/commit/ff4517e58337a83a26444905799e532e52d0d36a works fine for me.

usineur, 2017-01-22:
I confirm I encounter the same problem with latest revision. When rebooting from Decrypt9, I get black screen too.

DarthVitrial, 2017-01-23:
Same issue here, sometimes I'll just get a black screen on booting. Even when I don't, booting seems to take longer than it did on the latest stable build.

Shadowxp13, 2017-01-23:
Same thing here, when it happens, i have to make up to five reboots for a sucessful boot

Moshifan100, 2017-01-23:
Works for me just fine on latest commit. If anyone needs proof I can provide it.

TuxSH, 2017-01-23:
Does 6e5987e3ca219a63e0ff0f897309df0e8c534046 work for you?

Shadowxp13, 2017-01-24:
@TuxSH [6e5987e](https://github.com/AuroraWright/Luma3DS/commit/6e5987e3ca219a63e0ff0f897309df0e8c534046) Has the same issue for me (New 3DS XL A9LH Latest Firmware)

TuxSH, 2017-01-24:
So it works, eh?

What about the latest commit I just pushed, then?

Shadowxp13, 2017-01-24:
@TuxSH Your latest commit shows my splash screen and can enter the config and the chainloader,this didn't happen on the previous commits. But after the splash screen, chances of sucessfully booting into HOME MENU are random.

TuxSH, 2017-01-24:
To be sure, 6e5987e3ca219a63e0ff0f897309df0e8c534046 and f03e232b90ed3e2e8545c5abb4c489ca4f1e0b7d work fine, do they?

TuxSH, 2017-01-24:
And 61ecd9a61781e66005459917b511dc089041fc28 ?

Moshifan100, 2017-01-24:
yes they all work fine for me.

Shadowxp13, 2017-01-24:
6e5987r and f03e232 works for me. But 61ecd9a doesn't.

TuxSH, 2017-01-24:
Hm, I need to determine the cause of the issue though I'll solve it

Shadowxp13, 2017-01-24:
I'm making some tests when using a pin with the hide option enabled

mshipman, 2017-01-24:
I was having this issue with all commits since f03e232 when using the unofficial hourlies at https://astronautlevel2.github.io/Luma3DS/

However, once I compiled Luma myself, I encountered absolutely no more problems on any recent commit. (I have a regular x86_64 Linux box and a Raspberry Pi set up to build automatically for testing, now)

I'm just wondering if this issue is a build environment problem, rather than an issue with a recent commit.

The latest commit **Works for Me (tm)**

usineur, 2017-01-24:
@mshipman Are you using the devkitpro r46 or a previous one ?

mshipman, 2017-01-24:
@usineur I'm using r46 on both systems (and the latest commit of libctru).

usineur, 2017-01-24:
@mshipman ok just like me, I compiled it myself too, and it's not working properly... M'kay

mshipman, 2017-01-24:
@usineur I'm trying to think of anything else I could've done differently. I did compile the latest makerom and armips from source, as well, rather than using any precompiled binaries for them.

mshipman, 2017-01-24:
Update: I've been able to reproduce the failed boot on my N3DS XL on the latest commit. It's very rare, but there's definitely something going on... I feel like I'm in the Twilight Zone!

Today's build: occasional failed boot ---
[Luma3DS_61ecd9a61781e66005459917b511dc089041fc28.zip](https://github.com/AuroraWright/Luma3DS/files/727556/Luma3DS_61ecd9a61781e66005459917b511dc089041fc28.zip)

Yesterday's build: I couldn't get it to hang on boot ---
[Luma3DS_6f56a9bfe95b2d2b543b51006290c95e04932c37.zip](https://github.com/AuroraWright/Luma3DS/files/727561/Luma3DS_6f56a9bfe95b2d2b543b51006290c95e04932c37.zip)

(I'd be curious to see if anyone else who is having the issue can successfully run my build of 6f56a9b without boot troubles)

Shadowxp13, 2017-01-24:
The issue is very ramdomic, so it is kinda hard to say that there is no more issues.
I've compiled it myself ang got the same issue

usineur, 2017-01-24:
@mshipman I also use self-compiled versions of makerom and armips (last revisions or almost). If I have time later that evening, I will give a try to your build of 6f56a9b

TuxSH, 2017-01-24:
But https://github.com/AuroraWright/Luma3DS/commit/6e5987e3ca219a63e0ff0f897309df0e8c534046 reliabily boot when compiled from source, right?

TuxSH, 2017-01-24:
Nvm, please test 6c5f6ac475931579f4f5432af963388b5603a8ee

mshipman, 2017-01-24:
@TuxSH The new commit 6c5f6ac seems to be booting reliably so far. I will continue to test it for a while and update if I'm able to reproduce the issue from before.

Moshifan100, 2017-01-24:
No problems for me on any of the commits, but I compile it myself with the latest version of everything. Maybe it's people using prebuilt versions idk

TuxSH, 2017-01-24:
Does 07101c053a468e0d028b134911383a5dff533bfd work too?

usineur, 2017-01-24:
@TuxSH 07101c0 is ok for me. Rebooting from "Homebrew Launcher" and "Decrypt9" several times without issues

Shadowxp13, 2017-01-24:
Rebooting into Windows for testing  07101c0

Moshifan100, 2017-01-24:
@TuxSH 07101c0 works fine for me, rebooting from Decrypt9 many times and no problems.

Shadowxp13, 2017-01-24:
@TuxSH 07101c0 Works! 
ef2e008 works too.

closed 2017-01-24 by TuxSH