LumaTeam/Luma3DS #346: black screen when using latest commit

Cysime, 2017-01-23:
i got a problem that when i use the latest release, then i cannot launch Pokemon Sun(cart & cia)
it will work on the latest commit, but when i reboot my New3DSXL(USA),i will black screen when booting.
i FOLLOWED the FAQ about this issue, but it does't work.(NO "svcBreak" in the "loader",just a black screen)

TheBigQuack, 2017-01-23:
same problem cannot boot reverting does nothing

TheBigQuack, 2017-01-24:
guess we just have a bricked system lol

Cysime, 2017-01-24:
@TheBigQuack i can boot my system by reverting the bakup, but cannot launch the Pokemon Sun... pretty strange

TheBigQuack, 2017-01-24:
well i had a backup copy of the sd card and that fixed it

urherenow, 2017-01-24:
Special cercumstanses? Region Emulation? I'm imagining a corrupt file caused by an open for reading and failing to close it again

TheBigQuack, 2017-01-24:
once i recovered i tried update again worked once i manually powered off after update

Pitel, 2017-01-24:
Duplicate of #345?

TuxSH, 2017-01-24:
Duplicate

closed 2017-01-24 by TuxSH