LumaTeam/Luma3DS #347: [Feature Request] Patch HTTPs to HTTP

HexOverflow, 2017-01-24:
Having this would allow for Nintendo Network custom servers.

james-d-elliott, 2017-01-29:
I would have thought the only way to bypass the https for custom servers would be modification of the cert chain, patching https to http seems too broad.

kitling, 2017-05-30:
I agree with @james-d-elliott . A global https -> http would be too broad and only useful for development. Speaking of which, @SciresM made a [patch](https://github.com/SciresM/3DS-SSL-Patch) you can use that disables CA checking, preforming much the same task.

Now, it could still be worth it to actually add our own CAs to the system, but we could also just patch the URLs to the custom server's URL, and use the existing CAs. Now, if certain popular CAs are not supported by default (*cough* Let's Encrypt *cough*) we should find a way to support these.

I don't know. Would we rather have custom CAs + DNS or stock CAs and (minimal) patches for every game/service? See [here](https://github.com/chaoskagami/corbenik/blob/master/external/contrib_patch/mysterymachine.pco) for an example of the latter.

Phalk, 2017-06-05:
I would prefer Custom CS + DNS. Patching games is always a hassle.

kitling, 2017-06-06:
@Phalk The patches required are extremely simple. Find the string corrosponding to the URL being patched in the ROM/in memory.

On the other hand, Custom CAs + DNS could be "easier". But that leaves your system just as open as SciresM patch, if only to specific people who have the control over DNS + CA. Think of someone creating a custom server, distributing a custom CA and DNS server, and grabbing your credit card information next time you use the eshop.

It should be evident which answer to that question I am currently leaning towards. >_>

That said, I'd still like to get Let's Encrypt working as a trusted CA. Even better, just replace the CA's with, say, Firefox's default list. So I'd still like to see work done on that front. Not that it hasn't already been done (just remembering conversations on IRC), I guess, but it hasn't been well documented yet.

Hikari-chin, 2017-06-08:
If you want to do this, you can use [SciresM's patch](https://github.com/SciresM/3DS-SSL-Patch), but this will not be integrated into Luma itself because it is inherently insecure. Potentially, anyone could intercept your login details, credit card information, etc. Closed.

closed 2017-06-08 by Hikari-chin