LumaTeam/Luma3DS #349: [Feature Request] 2.1 support

monkeyman4412, 2017-01-27:
just for the heck of it. I know call me crazy.

THEDARKK, 2017-01-28:
Yes 2.1 support new3ds :)

monkeyman4412, 2017-01-28:
well o3ds for now. keep in mind 2.1 is only for  old3ds. so your asking something a little harder then just giving support. However that would be cool.

Margen67, 2017-01-28:
But why?

monkeyman4412, 2017-01-28:
my quote "just for the heck of it. I know call me crazy." you answer lies there.

Mrrraou, 2017-01-28:
but isn't it already supported anyway lol

monkeyman4412, 2017-01-28:
no it isn't

Mrrraou, 2017-01-28:
how so, that's not what i remember lol

monkeyman4412, 2017-01-28:
it isn't. what console are you on. old3ds. or new? if new then no. if old Im unsure.

Mrrraou, 2017-01-28:
why does it even matter

monkeyman4412, 2017-01-28:
well first off they are different. second off n3ds never had the firmware 2.1. Meaning by default n3ds is somewhat incompatible and needs more patches than old 3ds to work.  And if you did a ctr transfer to 2.1. you will recive a ctrmount error. meaning that it isn't compatible for n3ds. And also she never  said 2.x was ever supported only 4.x and above. so if you have a old 3ds. and have a system backup. go ahead and ctr 2.1. if it doesn't work then yah, was not supported, and you recalled incorrectly.

Mrrraou, 2017-01-28:
that's some high quality bullshit you're telling me there

monkeyman4412, 2017-01-29:
I can show you. I'm not lying. So let me guess that it worked for you. which means you have a old 3ds. which means by  default that works. HOWEVER, n3ds doesn't work. So I'm I right on I'm I wrong about you owning a old 3ds. (see this the reason I asked because now you accusing me of lying, when we don't even know if we have the same version of said device )

Mrrraou, 2017-01-29:
i have a n3ds, you did great, good job

monkeyman4412, 2017-01-29:
did you even test? did you go and do a ctr transfer to 2.1

Mrrraou, 2017-01-29:
why would I need to do that... what you're asking for isn't even 2.1 support but support for o3ds ctrnand encryption on n3ds for Luma. and i think that should be guessed with the ncsd header or something...

monkeyman4412, 2017-01-29:
But then again still. We are asking n3ds support for 2.1. And that just could be a step necessary. HOWEVER. n3ds is still technically unsupported. 2.1 was released in 2011. before n3ds release (so not even Nintendo would have support for the darn thing.). Also For n3ds users, we receive screen issues. such as colors rapidly changing. And if you were doing this without luma and/or arm9 installed. going and using anything with settings  (I'm more specifying to storage)  the 3ds crashes. you could be right that you just need encryption support for old 3ds. However my request was for 3ds or all 3ds to have support for 2.1. So in a fashion, your correct, but I'm also correct.

piratesephiroth, 2017-01-29:
Isn't this supported since commit https://github.com/AuroraWright/Luma3DS/commit/3076d56973e6e05707e0094c3f82b76dcab20d16 ?

monkeyman4412, 2017-01-29:
n3ds is not supported however, Old may. but n3ds isn't

Margen67, 2017-01-29:
O3DS firmware on N3DS isn't supported by Luma

Mrrraou, 2017-01-29:
don't expect anyone to try to fix these issues with the 2.1 firmware for n3ds, the screen color issues aren't even related to the FIRM but to the GSP sysmodule
and really no one gives a fuck

i think the only thing that is going to be done will be the o3ds ctrnand encryption support on n3ds for Luma itself and 2.1 will boot. but you'll definitely have fun with bricks (and "permanent" ones too)

AuroraWright, 2017-01-29:
2.1 bricks (possibly permanently) the N3DS in sleep mode...

closed 2017-01-29 by AuroraWright