LumaTeam/Luma3DS #350: Region free fails with Maido Hanahuda

gingerbeardman, 2017-01-27:
I get a black screen when launching this game. I have to load it through Hans to play it. 

Game is Japanese eShop only. 
Listed as Maido Hanahuda
I call it Maido Hanafuda (毎度 花札)

https://m.youtube.com/watch?v=T729jUO-4EI

basvijfwinkel, 2017-02-06:
Have you tried adding a Japanese settings file to the 'locales' folder? 
I believe you need to add something like 'JPN JP' and name the file the same as the game id.
Also do check that the region-free option is disabled by default.

gingerbeardman, 2017-02-06:
Region free works for all my other non-PAL games (only 6 in total)
Region free is enabled in Luma settings.

I've tried 00040000000B6700.txt in the /luma/locales folder, still no joy.

HugaShire, 2017-02-07:
use luma locales cia

gingerbeardman, 2017-02-07:
@HugaShire can you provide a link? Not sure what CIA your referring to? 

I used LumaLocaleSwitcher to setup the locales file but the game still doesn't run.

MelonGx, 2017-03-06:
/luma/locales/00040000000B6700.txt ONLY WORK FOR 6.6 release and earlier.

If you're using NIGHTLY then please use /luma/titles/00040000000B6700/locale.txt.

ericjwg, 2017-03-06:
https://github.com/ericjwg/LumaLocaleSwitcher/releases

Margen67, 2017-04-04:
@gingerbeardman Has this issue been solved?

gingerbeardman, 2017-04-05:
@Margen67 I use the binary available above from @ericjwg

Margen67, 2017-04-06:
ericjwg's fork is but the paths it uses are not. Luma nightly changed the paths so this is no longer a Luma issue, only a LocaleSwitcher one.

gingerbeardman, 2017-04-06:
@Margen67 then you know more than I do about whether it is fixed or not, sorry

closed 2017-04-17 by AuroraWright

gingerbeardman, 2017-06-05:
Just got around to testing this again, and the game will still not boot

I am using `/luma/titles/00040000000B6700/locale.txt` with contents `JPN JP`

patching is on in Luma (v8 with boot9strap)

any ideas?

gingerbeardman, 2017-06-05:
OK, this game will only run on the latest Luma+B9S setup with the following New 3DS CPU settings:

Off= YES
Clock = YES
L2 = NO
Clock+L2 = NO

Thankful I have found a solution!