LumaTeam/Luma3DS #354: New3DS Brick? (Black screen/shutdown)

coc4tm, 2017-02-07:
3DS Model : New3DS
3DS SysVer : 11.2E
3DS CFW : Luma3DS
3DS CfwVer : latest commit build

Hello , my 3DS won't boot with and without SD Card (i have installed A9LH to CTRNAND using plai's guide) . That's a new problem . I guess the problem come from the SD Card  but i re-formatted my SD Card to MBR/FAT32 and the problem is always here.

I tried these A9LH payloads : 

- Luma 3DS 6.6

- Luma 3DS automatic last commit build

- Decrypt9WIP

Black screen in any case .
I cant' boot Luma3DS config menu / Ctrboot menu

Margen67, 2017-02-09:
What did you do before it bricked?

jcmejia, 2017-02-09:
No haven't done it before I only have 2 n3ds one on 9.2 and the other at
latest firmware so yeah I only have one I can mod I just don't want to risk
it

jcmejia, 2017-02-09:
But I do have another question I was told if my SD card has a certain
partion table it can prevent the random bricks when installing a9lh

coc4tm, 2017-02-10:
@Margen67 

Absolutely nothing .
That's pretty weird

GizmoTheGreen, 2017-02-14:
not sure if it's related but I updated my n3ds's to 11.3 and they worked just fine with multiple reboots then suddenly they stopped booting(with black screen)

removing firm.bin in luma folder fixed it (was left over from when you needed it to run ntr) also worth trying is removing your luma config file?

coc4tm, 2017-02-16:
I will try this . thanks

Gildedboy, 2017-02-27:
try to change the dev features in luma configuration menu from "off" to "errdisp", if is a black screen error it will tell the type of error

coc4tm, 2017-02-27:
@Glidedboy

I can't access Luma configuration menu

Margen67, 2017-02-27:
If you have the SD card inserted but there's no payload (arm9loaderhax.bin) on it does the 3DS turn off?

coc4tm, 2017-02-28:
@Margen67 yes
I haven't tried yet the solutions from @GizmoTheGreen

infinity5, 2017-03-04:
I once had the same problem, but I fixed it by removing everything in the "sd://luma" folder and then getting the latest "arm9loaderhax.bin". I then booted into the config menu and everything worked!

coc4tm, 2017-03-04:
Solutions from @GizmoTheGreen and @infinity5 . _Je suis dans la merde_

AuroraWright, 2017-04-17:
Luma can't cause bricks, try another SD card, if it fails it's a hardware failure of some kind

closed 2017-04-17 by AuroraWright

coc4tm, 2017-04-21:
Yes , it's not related to Luma3DS but to the SD Card corruption.
I'll send back my 3DS to Nintendo.
Thx