LumaTeam/Luma3DS #355: Gba virtual consoles screen issues

jcmejia, 2017-02-07:
Gba virtual consoles have screen issues on luma3ds 6.6 firmware 11.3.0-36u

Margen67, 2017-02-07:
What is the allocation unit size?

jcmejia, 2017-02-07:
128gb

jcmejia, 2017-02-07:
On new 3ds

jcmejia, 2017-02-07:
It was working fine on 11.2 on luma3ds

piratesephiroth, 2017-02-07:
what's even a screen issue?

jcmejia, 2017-02-07:
After I updated Luma to 11.3 it started having issues

jcmejia, 2017-02-07:
Try a gba virtual consoles on Luma3ds on firmware 11.3

jcmejia, 2017-02-07:
The resolution is messed up I'd post a pic of it but I don't know if I could

cheatfreak47, 2017-02-07:
@jcmejia

>128GB Bingo, I've found the issue.

Basically, GBA games sometimes have an issue where the screen is mapped improperly when using a card that is too slow. **This is not a Luma issue, this happens even on stock systems.** 128GB cards tend to not be as fast as smaller size cards, which may be the cause here.

As @Margen67 was asking, the allocation unit size (sometimes called cluster size) can also have an impact on this, so your only hope of fixing this issue is to change it.

The best course of action is to backup your entire SD card to your computer and format the SD card using this [tool](https://cdn.discordapp.com/attachments/265713680560160768/278579320287395840/guiformat.zip).
When formatting set the Allocation Unit Size to "32768", and make sure you pick your SD card and not some other important drive, and from there, if you are lucky, the GBA issue will disappear, otherwise tough luck, it's nothing we can just fix.

In the meantime, this is not only a non-luma issue, but also a duplicate of an issue that was already closed, and thus this should also be closed.

shinji257, 2017-02-07:
Duplicate: https://github.com/AuroraWright/Luma3DS/issues/142

jcmejia, 2017-02-07:
The cluster size I don't know but I will check it later but I think it's a
32kb cluster

jcmejia, 2017-02-07:
Besides that it was working fine until I updated Luma to
11.3

cheatfreak47, 2017-02-07:
Look, I know exactly what your issue is and I've done research into it personally, and I'm telling you the issue is definitely related to SD card speed **and has nothing to do with the update, nothing to do with Luma, and nothing to do with hacking either.**

Nintendo hasn't touched AGB_FIRM since they added it with the ambassador program back when it was introduced.

You probably just never encountered the issue before, because it's not entirely consistent.
It happens seemingly randomly with different SD card sizes and cluster sizes.

jcmejia, 2017-02-07:
So what do you recommend

cheatfreak47, 2017-02-07:
What I said above, reformatting the card as FAT32 with 32,768 bytes per allocation unit (cluster size) and hoping the issue goes away.

If it doesn't either deal with it or get a 64GB SD card instead.

Margen67, 2017-02-08:
Or using a GBA emulator

StateOfException, 2017-02-09:
Obviously, this isn't a Luma issue- but I have noticed that this issue only crops up in certain titles.

Kurukuru Kururin, for example, exhibits the issue- whereas, say, FF Tactics Advance doesn't. Might explain why @jcmejia didn't notice the issue previously - if they're trying a VC inject they hadn't used prior to upgrading Luma version, and it happens to be one of the unlucky ones, they wouldn't have any idea what prompted the change. 

If you'd like some confirmation, try some other games.

jcmejia, 2017-02-09:
Actually it was happening with sonic advance 2 and 3 on emunand but not on
system nand

jcmejia, 2017-02-09:
But I did notice a slight fix by pressing the home button a couple of times

StateOfException, 2017-02-09:
> But I did notice a slight fix by pressing the home button a couple of times

...Strange. This does actually have an effect- rapidly tapping the Home button shifts the output up by a certain interval on each press. Seems to take place during the transition time between the back-to-Home-Menu dialog being called and its display animation completing displaying. 

This is the result:

[image]

By repeatedly doing that, it's possible to make the output loop all the way around, making it output correctly. Takes about a minute or so of tapping, tops.

Peculiar bug.

jcmejia, 2017-02-09:
Yes that's exactly what I did but for me it was less than a minute

Margen67, 2017-02-09:
This is why you shouldn't use emuNAND.

StateOfException, 2017-02-09:
It's not just an EmuNAND issue, @Margen67 - I can reproduce it from SysNAND.

Not really a big deal, I suppose. Would be nice to have some kind of way of preventing it happening, but that moves well out of bug report territory into feature requests.

jcmejia, 2017-02-09:
So should I use system nand

StateOfException, 2017-02-09:
@jcmejia This isn't really the correct place to ask for general tech support. I recommend asking on [/r/3dshacks'](http://reddit.com/r/3dshacks/) Q&A thread. 

(But, yes- you should transfer your EmuNAND setup to SysNAND. Ask in the Q&A thread if you're not sure how that's done.)

urherenow, 2017-02-09:
So... yes, the screen scrolling with the home button IS a bug. 

But for future reference, VC titles from emunand pose a problem on a variety of setups (hacky reboot patches required to work AT ALL) so it is best practice to use sysnand for VC. The latest guides for a9lh don't even bother telling you to make an emunand anymore.

Margen67, 2017-02-09:
>it is best practice to use sysnand for VC.
It's best practice to use sysNAND for everything not just VC unless you're a developer, want to bypass the 300 title limit, or region change. Otherwise it's just a waste of SD card space.

jcmejia, 2017-02-09:
I will probably stick with emunand cause I haven't installed a9lh before I
heard it's very risky due to random bricks I'm actually more skilled with
wii/wii u modding I'm actually still learning on 3ds modding I'm actually a
fast learner when it comes to modding besides that thanks for letting me
know about a9lh I might install it sometime in the near future.

Margen67, 2017-02-09:
@jcmejia All bricks are due to user error.
As long as you follow the guide you won't brick. https://3ds.guide/

closed 2017-03-06 by TuxSH