LumaTeam/Luma3DS #356: Black screen on boot, Data Abort (menu)

TwoSideA, 2017-02-09:
Running on Luma 3DS, JP N3DSXL, firmware is around 11 or so (haven't been checking). 

Once I boot the system, there is a faint image of my home screen then it flickers to black. It flickers back in forth in the span of 2-3 minutes. Once I hold the power button, the power off prompt screen is displayed normally as it does. I did not attempt to update my firmware or Luma 3DS prior to this, I just changed themes through CHMM2 and once it rebooted, that's where the problem started.

Here is my crash dump. 
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/762371/crash_dump_00000000.zip)

Margen67, 2017-02-09:
What entrypoint are you using?

jcmejia, 2017-02-09:
Menuhax

jcmejia, 2017-02-09:
CFW is luma3ds updated to latest firmware 11.3

TwoSideA, 2017-02-09:
I'm assuming Menuhax, since I followed Plailect's guide.

Margen67, 2017-02-09:
Plailect's guide has you install A9LH at the end not menuhax.

You should switch to A9LH http://3ds.guide/

TwoSideA, 2017-02-09:
My apologies. But I do have A9LH installed as I finished going through the guide. I did about 6-8 months ago so I don't quite remember what is what.

TwoSideA, 2017-02-09:
So I found a fix. After booting up my 3DS, I just went ahead and remove the microSD card and the screen reverted back to normal. Then I put it back in, all my files (FBI etc) came back on screen and now I can use my 3DS normally.

Weird.

closed 2017-02-09 by TuxSH