LumaTeam/Luma3DS #357: Increased Clock & L2 Cache on per-title basis?

theKKCD, 2017-02-10:
Because some games do not work with Clock or Clock+L2, I was thinking that this could be implemented on a per-tile basis.

In my model, you would be able to set a "default" state, and then use titleIDs in txt files (similar to how locales work now) in order to switch the mode for that particular game. For example, I would set my default to Clock+L2, and then disable that for SMT and others rather than having to go through Luma Config menu every time I want to play that game.

There could also be an app on the home menu (similar to LumaLocaleSwitcher) to change this from the 3DS itself.

Shadow322, 2017-02-10:
Use NTR then u can turn it on and off at will

theKKCD, 2017-02-10:
I didn't know you could do that. Thanks!

I think it would still be better to have it persistent. And for some reason I cannot use sleep mode while running NTR or my 3DS will never wake up.

sora10pls, 2017-02-10:
this has been suggested at least 20 times I think lmao

probably not gonna happen

vaguerant, 2017-02-11:
You can see TuxSH and AuroraWright's responses to this in #25 and #70, respectively.

theKKCD, 2017-02-12:
So it's just not possible then? Well that's too bad. Closing.

closed 2017-02-12 by theKKCD