LumaTeam/Luma3DS #361: Black screen

DeathWizard252, 2017-02-13:
I have just started Luma3DS for the first time on 11.3.0.36 and then I boot it up, it comes to a black screen. I don't know how or why this is, but I followed the instructions.

Margen67, 2017-02-13:
Do you have A9LH?

DeathWizard252, 2017-02-13:
Yeah, I do.

Margen67, 2017-02-13:
Can you get into the options menu by holding select while starting the 3DS?

DeathWizard252, 2017-02-13:
No, I can't.

Margen67, 2017-02-13:
Can you boot into recovery mode by holding A+R+L+Up(DPad) while turning on the 3DS?

DeathWizard252, 2017-02-13:
No.

Margen67, 2017-02-13:
What if you take out the SD card?

DeathWizard252, 2017-02-13:
Can't access it at all because hombrew is based off of something on my SD card

DeathWizard252, 2017-02-13:
Wait, nevermind. I thought you had to boot the 3ds, and you had to boot Luma. I'm still having trouble, as it says 'An error has occurred while loading the payload.'

closed 2017-02-13 by DeathWizard252

Margen67, 2017-02-14:
Can you take a picture of the error?