LumaTeam/Luma3DS #363: Can't enter configuration menu

Assaro, 2017-02-15:
Hey, just installed Luma, but the configuration menu didn't show up. Any reasons?

Margen67, 2017-02-15:
Can you get into recovery mode by holding A+R+L+Up(DPad) while turning on the 3DS?

Assaro, 2017-02-15:
Getting this:
[PANIC] Target payload not found:  /luma
Press any key to shutdown

Margen67, 2017-02-15:
Can you take a picture?

Assaro, 2017-02-15:
[image]
there you go

Margen67, 2017-02-15:
Try turning off "use custom path"

Assaro, 2017-02-15:
?How you do that?

Not considering myself a pro

Margen67, 2017-02-15:
Hold select while turning on the 3DS.

If that doesn't work take out the SD card and delete the config and path.txt in the luma folder.

Assaro, 2017-02-15:
[image]
Now I am here

Assaro, 2017-02-15:
Alright i was able to reset it back to where it started. I'll try again and see whether i will work

Margen67, 2017-02-15:
Are you using menuhax or A9LH?

Assaro, 2017-02-16:
Used A9LH. Have made an unknown mistake, old file, etc.
Did it again and it worked just fine. Loving it :)

closed 2017-02-16 by Assaro