LumaTeam/Luma3DS #364: O3DS/2DS Freezing

BARNWEY, 2017-02-16:
Before writing this issue, I'd like to say that I am not the one having this issue. It is actually @Droople on GBATemp. He has been having issues with his Home Menu freezing after setting up A9LH and updating to 11.3, and it doesn't seem like there is any issues with his setup...
Here's a video:
https://youtu.be/lzMAIp4MKVw

Please post on this thread if you have any suggestions for him:
http://gbatemp.net/threads/2ds-frozen-when-quiting-from-some-apps-to-home-menu.461175/#post-7101316

Thank you for your time, and have a nice day!
P.S. His setup is:
2DS on 11.3 running Luma w/ A9LH v3
He has a class 10 SD card, and has his payload and files on the root of his SD...

Margen67, 2017-02-16:
Are any memory mode changes being used?
Are dev patches enabled?
Are there any files like firmware.bin in /luma?

droople, 2017-02-16:
No (what's this?),
No (what's this?), sorry I'm a noob in CFW.
No, only config.bin
All the steps followed and only followed from 3ds.guide

FateForWindows, 2017-02-16:
Don't have this issue, try the latest hourly.

BARNWEY, 2017-02-17:
It seems like he fixed the issue after formatting.... Thanks to everyone who helped @Droople fix his 3DS!

closed 2017-02-19 by TuxSH