LumaTeam/Luma3DS #365: Using Gateway linker

TheZeldaBoss, 2017-02-17:
Hello,

My parents gave me a new 3DS XL last christmas, and I wanted to use my gateway linker on it. As it didn't have the good firmware, I installed Arm9LoaderHax and Luma3DS on it in order to operate a downgrade. Now, my 3DS has the correct firmware, but I get some issues when I want to use my linkers. I added a payload to load Gateway, but I can't get the menu, and I used to be able to use my DS linker, but not anymore...

Are there a way to remove the whole hack while keeping the actual firmware ?

Thank you !

cheatfreak47, 2017-02-17:
Why would you bother with a Gateway when you already have a fully functional CFW? 

Just install your games as CIAs or use something like freeShop or whatever.

cheatfreak47, 2017-02-17:
First of all, you would only need "dozens of cards" if you are using tiny SD cards or you wanted to catalog every 3DS game ever made. 

and for that matter, wouldn't you _also_ need "dozens of cards" if you are using a Gateway? GW doesn't magically remove the requirement for SD storage space. It's the same stuff being used in a crappier way.

If by "DS linker" you mean the DS flashcart thing, then this guide's page should help fix that.
https://3ds.guide/troubleshooting

TheZeldaBoss, 2017-02-17:
And then, are there a way to : 
1) convert .3ds files to .cia easily ?
2) Use saves obtained from gateway as ones for each game installed ?

If yes, I'll forget gateway. Else, I'll wait a bit.

cheatfreak47, 2017-02-17:
1. You can do so with decrypt9 right on the 3DS if you want (though that's kind of a waste of time, when most games are avaliable in CIA format anyway)
2. You can do that by getting a 3DS into Gateway mode, and using JKSM to dump the saves from the Gateway one at a time, and then restoring them to the digital version later.

closed 2017-02-27 by TuxSH