LumaTeam/Luma3DS #366: MSET problem?

macraix, 2017-02-18:
Hello, I was wondering if someone can help me. I'm trying to make Luma for MSET to work but I can't make it work. I've followed a lot of guides and nothing works :( someone can help me? Maybe I'm missing something?
I've put the necessary files to work but when I press the "Nintendo DS Profile" option my 3DS freezes and then says "An error has occurred"

Margen67, 2017-02-18:
Why are you using mset?

macraix, 2017-02-18:
Sorry but if you don't know how to help me please don't comment, I like MSET and that's it.

Margen67, 2017-02-18:
You should follow this guide: https://3ds.guide/

urherenow, 2017-02-19:
What's your FW version and how are you installing mset?

Margen67, 2017-02-19:
You shouldn't use MSET. You should install A9LH using the guide.

macraix, 2017-02-19:
@urherenow I'm installing it with Gateway Blue Card and I'm on 4.3

Margen67, 2017-02-19:
@macraix Don't use MSET follow this guide. http://3ds.guide/

macraix, 2017-02-19:
@Margen67 Oh god.... I ALREADY HAVE OTHER 3DS WITH A9LH AND OTHER 3DS WITH MENUHAX SO CAN YOU PLEASE STOP TELLING ME THAT? IT'S MY DECISION TO HAVE MSET -.- If I wanted to have a9lh I WOULDN'T ASKED THIS QUESTION FIRST, so please stop commenting. thanks

Margen67, 2017-02-19:
Why would you want MSET or Menuhax?

Mattery, 2017-02-19:
Well i can't understand why would you want to use mset but did you organized the files exactly as it's shown here? https://github.com/AuroraWright/Luma3DS/wiki/Installation-and-upgrade 

By the way what version are you on? you can't open an issue without giving any information and i recommend you to use a9lh anyways, mset is deprecated by now.

macraix, 2017-02-19:
@Mattery Thanks, I did that exactly and nothing happens. Sorry for not saying my version, my sysNAND is on 4.3. I'm using Gateway blue card to install the mset.
As I've said before, I have all of the files correctly. I don't know whats wrong :/

Margen67, 2017-02-19:
@macraix Why do you want MSET?

Xtheman166, 2017-02-19:
Like Margen said, why do you want MSET? MSET is old and shitty, and you are fine installing A9LH as you have stated you have installed it on your other 3DS

macraix, 2017-02-19:
Sorry if I'm being an ass but please stop telling me what can I do or what can't I do, I like MSET. I don't like a9lh and that's it, so if anyone can help me I would be grateful, if there's nothing I can do to run it I will wait for GW to release their update so I can update my emuNAND to 11.3.

Margen67, 2017-02-19:
What don't you like about A9LH?

Mattery, 2017-02-19:
Btw it doesn't seem that the gateway mset installer was designed for 4.3 so it could perfectly be a gateway issue and nothing to do with luma, you should ask this in gbatemp not here in my understanding.
Btw, there is no reason at all for using mset, don't expect anyone to help because you're probably the only one trying to use it.

TuxSH, 2017-02-19:
Luma3DS's arm9loaderhax.bin is at the root of your SD card _even if_ you're not using a9lh, right?

Mattery, 2017-02-19:
https://github.com/AuroraWright/Luma3DS/issues/58 take a look at this @macraix

macraix, 2017-02-19:
@Mattery Ohhh.. So the problem is that it doesn't support any firmware below of 7.x, right?

Mattery, 2017-02-19:
Not officially at least so if it doesn't work after the base installation bad luck.I doub't someone will start checking this in an ancient fw.

Mattery, 2017-02-19:
And EmuNAND and GW aren't necessary at all right now so you should move to 11.2/3 Sysnand+A9LH if you want support and total compatibility.

Shadow322, 2017-02-19:
@macriax Don't be sorry that Margen67 is always an asshole to anyone not using a9lh :)

macraix, 2017-02-19:
I own a Gateway myself so I don't need to use another CFW, thanks for your help @Mattery

pbanj, 2017-02-19:
Why are you posting on lumas issue tracker?

urherenow, 2017-02-19:
@pbanj because Luma includes Luma.nds when you build it, but mset is broken?

urherenow, 2017-02-19:
Oh, ok. Damn, I must be getting old...

NutymcNuty, 2017-02-20:
crrriiiinnnngeeeeeeeeee
why are you using mset still aleast use menuhax its like .5x better

Zewia, 2017-02-20:
https://3ds.guide/

Follow it.

pokecraft98, 2017-02-20:
How is this issue still up? Anyway follow https://3ds.guide

closed 2017-02-26 by TuxSH