LumaTeam/Luma3DS #367: Region free not working on latest hourly

daothanhduy305, 2017-02-20:
For example, the game I'm having trouble with is Nintendogs and cats: Golden retriever and friends.
My system is 11.3, with latest luma hourly.
The game works with luma 6.6 stable release (that means I put the right txt file under luma/locales), but doesn't work with the latest hourly (I enabled the option game patching in luma already).

Nyquis, 2017-02-20:
In the nightlies, the folder structure for patches, locales and romfs has changed. You can take a look at it in this [commit](https://github.com/AuroraWright/Luma3DS/commit/b5336c81cc82b6c5e8115249342beb5b065cdce9).

You may use the new luma _locale switcher rvn_ which knows the new strucutre. It can be downloaded from _TitleDB_ in FBI.

daothanhduy305, 2017-02-21:
Confirmed! And thank you very much, and sorry for not looking into the commits/source at first hand

closed 2017-02-21 by daothanhduy305