LumaTeam/Luma3DS #369: Suggestion: don't display the chainloader menu if there is only one payload available

Siphonay, 2017-02-26:
Hi! Thanks for your work on Luma3DS. The new payload menu is very nice and is easier to use for people who use several payloads, but for the ones like me that only use one of them, it feels kind of cluttery to have  a menu with a single entry displayed and having to press A to boot into it. It's not a huge problem, but directly booting into a payload if it is the only one in the payloads folder would make it feel smoother.

Of course, it's just a suggestion and I can understand that you would not agree with me, or wouldn't want to implement it for any other reason.

Thanks for reading! Have a nice day.

cheatfreak47, 2017-02-26:
I agree- this should probably be added.

Nyquis, 2017-02-27:
I think as an option it could be nice. I would however disagree. The advantage of this menu is that it shows the button it is associated with as well as what the payload is before loading it. Second advantage is that on can refuse to load the payload.

pbanj, 2017-03-02:
I think the way the chain loader is now is good. You don't need to rename the payloads, it makes it more noob friendly. The amount of people who screw up the renaming of the payloads is really sad, this will stop that.

Committee, 2017-03-02:
Apologies but how do you open the chainloader menu in the nightly build?

tiduscrying, 2017-03-02:
@Committee, you press START before or after the splash screen (if you have one enabled and configured in the Luma menu). 

I was super excited for the payload menu when I saw the initial commits! Using the latest nightlies, I think it absolutely rocks. It isn't too big of a deal that it launches using start, but would it at all be possible to configure in the Luma menu what button launches the payload menu (I personally want to set it to "B")?

cheatfreak47, 2017-03-05:
I just like the idea that if there is _only one payload_ it'll just _launch it_ instead of making you _pick the only payload and then launch it._

urherenow, 2017-03-06:
You have to press a button to launch the menu anyway. Why don't you press the assigned button for your payload instead?

Siphonay, 2017-03-06:
@urherenow the button assigning system has been removed and replace by a menu that always open with the start button.

urherenow, 2017-03-06:
No it hasn't. It's in addition to. I can hold "down" when powering on and I immediately launch into Gateway. I build my own so there's no question I use the latest.

Nyquis, 2017-03-06:
I concur with @urherenow  All the latest nightlies allow one button direct boot up. Did you mistakenly forgot to rename your payloads?

Siphonay, 2017-03-07:
Last time I asked I was told there was no other option to launch payloads than the menu anymore. I either was misinformed or this was changed.

urherenow, 2017-03-07:
Never changed. It's been the same since way before there was a payload menu.

TuxSH, 2017-04-02:
You could use the chainloader w/o the menu, that is to say, with payloads bound to keys.

closed 2017-04-02 by TuxSH