LumaTeam/Luma3DS #371: fw update

pcs3rd, 2017-03-01:
is it safe to update from 11.2.0 to the newest firmware on emunand

Margen67, 2017-03-01:
Why are you using EmuNAND?

pcs3rd, 2017-03-01:
I originally had sysnad on a16gb card and ran out of space and couldn't get it to work after copying to a 32 gb card. so I have one of each, a 16gb sysnand and a 32gb emunand. just normally use the bigger one because it works. pretty much the same configuration, I come most of the config files.

pcs3rd, 2017-03-01:
* and I can backup all of my games

tiduscrying, 2017-03-01:
It is safe. Just for future reference, feel free asking questions like this on /r/3DShack's sticky thread or on GBATemp! The issue tracker for Luma isn't really the place to ask, it's for reporting issues!

Margen67, 2017-03-02:
If you're not already using A9LH you should follow this guide: https://3ds.guide/

closed 2017-03-03 by TuxSH