LumaTeam/Luma3DS #372: Longer boot up on 5081439f

Nyquis, 2017-03-06:
I currently have the impression that the boot up time on the latest nightly is visibly longer. What could it be linked with?

Margen67, 2017-03-06:
What entrypoint are you using?

Nyquis, 2017-03-06:
I am booting directly into sysnand. I have the a9lh + luma 3ds setup.

Margen67, 2017-03-06:
If you have any firmware.bin on the SD card then delete it.

Nyquis, 2017-03-07:
Last well working firmware was b1b81c8. I'm still on 11.2.  The difference
in speed is of about 5-9 seconds after the splash screen. The splash screen
has been configured to load after payloads.

Margen67, 2017-03-07:
Splash screens increase boot times.

Nyquis, 2017-03-07:
I know. However on both b1b81c8 and 5081439f. The same splash screen was configured to boot after payloads. As in my setup has not changed except for changing luma3ds build.

mshipman, 2017-03-08:
I too have splash screens enabled, and for any commit newer than b1b81c87f28baa43023ee024358999ce12d170bf rebooting from within an arm9 payload causes the system to hang on a black screen. Even after removing the battery, the system completely refuses to boot (boots up to a black screen) until I remove the SD card and revert to an older stable commit. After a successful boot, I can then re-update to the latest commit of luma and the system works normally until I once again try to chainload something like Hourglass9 and press start to reboot.

Lime Green N3DS XL - 11.3.0-36U - Luma + A9LH
(splash enabled before payload / Clock+L2 / booting SysNAND)

Nyquis, 2017-03-21:
Seems like the 0eab9127 nightly has fixed longer booting times.

closed 2017-03-21 by Nyquis