LumaTeam/Luma3DS #373: Code Sections usage causes game to crash 3DS

deadair3210, 2017-03-07:
I have used mystery machine patcher to patch the code.bin of the Pokemon games, but whenever I put the modified <titleid>.bin and enable language emulation and external code, the game tries to launch but gets to about half way through the nintendo animation, then crashes.

Dump files from crash available if needed

Running latest firmware, 6.6 Stable

rocknard, 2017-03-18:
Try update to hourly build and try (with Luma Updater).

deadair3210, 2017-03-19:
Will do when I get a chance, and report back

deadair3210, 2017-03-19:
Nightly works like a charm after I figured out the new folder structure. Thanks!

deadair3210, 2017-03-19:
There needs to be at least a notice somewhere though that external code is broken in stable though

closed 2017-03-19 by deadair3210