LumaTeam/Luma3DS #374: [Feature suggestion] Number crash dumps using RTC value.

sirocyl, 2017-03-07:
This provides a sequential, time-based identifier that's easy to avoid collisions on.

It prevents the situation where you have a large number of files, especially dump files, on a low-performing SD, and the `findDumpFile()` loop takes a long time.

TuxSH, 2017-04-02:
The problem is that IIRC the RTC doesn't contain the current date and time but an offset to what is stored on NAND. Currently there is no other way than using file names to sort the dumps per date of creation.

One could clean up their dump folders if they get too big.

sirocyl, 2017-04-03:
No, that's the point. It doesn't have to correspond to an actual timestamp, it is only being used to apply a name to the file, in a way that is sequenced roughly based on time.

You don't need to know the actual time and date, since the offset itself is a timebase.

TuxSH, 2017-06-05:
Meh.

closed 2017-06-05 by TuxSH