LumaTeam/Luma3DS #375: I'm not able to access to SysNAND

CorJoestar, 2017-03-08:
I tried to boot my New3DS by holding L button during the boot, but the screen remains completely black and nothing happens. I tried to set the autoboot in SysNAND, but when I try to boot my 3DS there is the same problem, but in this case when I boot by holding L button I boot EmuNAND successfully. 
I don't know what I have to do T^T

Margen67, 2017-03-08:
What entrypoint are you using?

CorJoestar, 2017-03-08:
Do you mean the Default EmuNAND location setting?
[image]

Margen67, 2017-03-08:
Are you using A9LH?

CorJoestar, 2017-03-08:
I'm not sure 'cause I've installed this CFW a long time ago, but I think so because in the root of my SD there are 2 files (arm9loaderhax.bin and arm9loaderhax.bin.bak)

Margen67, 2017-03-08:
Luma requires all entrypoints (including menuhax, MSET, etc) to use arm9loaderhax.bin

CorJoestar, 2017-03-08:
Mmm... so I have to install all of them, right?

Margen67, 2017-03-08:
Is there an arm9loaderhax.bin on CTRNAND?

If there isn't, then what happens if you boot the 3DS without an SD card inserted?

CorJoestar, 2017-03-08:
No, there isn't. When I try to Boot my 3DS it goes on SysNAND, but with the SD I can't do it :/

Margen67, 2017-03-08:
Try setting the home menu theme after turning the 3DS on without the SD card inside.

CorJoestar, 2017-03-08:
Done. Now?

Margen67, 2017-03-08:
If that fixed your problem now you should install A9LH by following this guide https://3ds.guide/

CorJoestar, 2017-03-08:
Okay, thank you very much! O.O
The next step makes me able to boot Luma again? 
So I try ^-^

CorJoestar, 2017-03-09:
Thank you so much! I solved all the problem! 😍

Margen67, 2017-03-09:
No problem. Now you can close the issue since it's been solved.

closed 2017-03-11 by TuxSH