LumaTeam/Luma3DS #378: (Feature Request) To only lock payloads and luma config not the boot

pokecraft98, 2017-03-11:
I would enable luma pin on my sd card but it means I have to unlock my 3ds every time someones use it. I don't want them to touch the payloads or luma config (i have a custom splash) but still being able to use it for games without me having to be there to turn it on. So maybe opitions like (Off, Boot, Payloads/Config).

Kaisogen, 2017-05-01:
I like this idea. I set up my friends 3DS with A9LH, and one time they booted into HG9. I was pretty scared they were going to do something stupid, even though I had a backup for them.

bladesdnm, 2017-06-05:
having the Rosalina menu doesn't help make things safer either. I wish I could disable it or go back to older luma.

Hikari-chin, 2017-06-09:
Duplicate issue; this has been mentioned in prior issues and it was decided against implementing it.

closed 2017-06-09 by Hikari-chin