LumaTeam/Luma3DS #379: Luma v6.6 doesn't load DS games on 2DS, O3DS TWL and AGB patch doesn't resolve issue.

PP-theSLAYER, 2017-03-12:
Both NDS retail carts and flash cart doesn't work.
(though my primary concern is Retail Carts right now)
[3DS carts work fine, so the cart slot isn't the problem]

It basically loads the cart and hangs on a black screen.
Removing the DS cartridge doesn't kick to the error message (it continues to hang on the black screen)
Wifi light still flickers once in a while.

The Clear Blue 2DS is on firmware 11.2.0-35E
I'm playing strictly on sysnand (no emunand)
Not on hourly build, 

Few details:
1. Not using Bootloader or anything (booting Luma3DS straight with A9LH)
2. Not using any form of path changer
2. Installed O3DS TWL and AGB patches, doesn't work.
3. Tried installing 0004800F484e4841 v0 from 3DNUS (version not spoofed), no dice
4. 3DS SD card format is FAT32

I've read about how others installed TWL and AGB to fix this, but using the O3DS files for that from the wiki didn't work for me.

Margen67, 2017-03-12:
You shouldn't install ***patched*** TWL or AGB firms. They will not work with Luma. https://3ds.guide/troubleshooting

Also delete any firmware files in the Luma folder.

PP-theSLAYER, 2017-03-12:
I downloaded the clean ones from Luma WIKI, but I'll try again with the ones you linked.

(also, my Luma folder only has the config.bin, and the payload for Decrypt9 (start button))

PP-theSLAYER, 2017-03-12:
That did the trick.

Looks like the TWL firm wasn't enough, but installing the other stuff fixed it.

Closing issue, thanks Margen67 for your help!

closed 2017-03-12 by PP-theSLAYER