LumaTeam/Luma3DS #380: Luma3DS would not boot via MSET after making a transition from rxTools (bootable via MSET)

dothackjhe, 2017-03-13:
Can anyone show me a screenshot of their memory card's root folder contents after successfully transitioning from rxTools to Luma3DS via the MSET route? 

I need to know which files are meant for Luma3DS and which ones are left for rxTools which likely affects the smooth transition i.e. Luma3DS booting via MSET. 

Attached is a screenshot of my O3DS XL MMC's root folder.

Note: The original files meant for rxTools were placed in the "rxTools Backup" folder as seen on the screenshot and not deleted.

[image]

Margen67, 2017-03-13:
Follow this guide: https://3ds.guide/

dothackjhe, 2017-03-13:
Can someone at least give an answer that does not point to that guide from the start? I know that guide exists, I just chose an easier route by asking a certain solution here. 

Following the guide I posted, it said about only replacing files to make a transition from rxTools to Luma3DS. That does not seem to be the case here.

Margen67, 2017-03-13:
>but I still prefer this method

Why?

dothackjhe, 2017-03-13:
I'm taking baby steps towards Luma3DS. A9LH is probably for later. I must boot to Luma3DS and familiarize on it first.

Margen67, 2017-03-13:
MSET is unsafe. It leaves you vulnerable to updates on sysNAND at boot and not to mention it's very inconvenient. You should immediately switch to A9LH.

dothackjhe, 2017-03-13:
I can go for Luma3DS and A9LH even on sysnand 8.1.0-19U, right? No updating required?

Margen67, 2017-03-13:
Yes.

dothackjhe, 2017-03-13:
I'm leaving this thread open in the meantime while I go through reading and comprehending the steps required in installing Luma3DS as bootable via A9LH.

Margen67, 2017-03-13:
Luma doesn't support MSET on 4.x.

dothackjhe, 2017-03-13:
I loaded Luma3DS.nds from my DS Flash Cart and saw two options with the ROP: patch for 4.X and 6.X. Given that my Sysnand is on 8.1.0-19U, I intuitively chose 6.X, yet it still did not fix the issue with Luma3DS booting via MSET.

Margen67, 2017-03-13:
You should just do the guide instead of wasting time with that.

dothackjhe, 2017-03-13:
Noted.

trbocode, 2017-03-17:
Assuming you didn't do it, try 4.X MSET

Margen67, 2017-03-17:
>Assuming you didn't do it, try https://3ds.guide/
fixed

jarupxx, 2017-03-17:
You use the Luma3DS.dat v6.3.1, Luma3DS will boot via the MSET route.

dothackjhe, 2017-03-18:
@trbocode 

I'll try that again and see how it works from there. Right now I'm still struggling identifying which files must be removed from the micro-SD Card after trying to make a transition from rxTools.

Margen67, 2017-03-18:
You don't need to remove any files after transitioning from rxTools except maybe any firmware.bin or othersimilar files. Otherwise it's just wasted SD card space.

dothackjhe, 2017-03-18:
@Margen67 Yeah, I was actually referring to those files but just do not know yet which to remove specifically.

Margen67, 2017-03-18:
@dothackjhe You shouldn't need to remove any files. Just follow this guide https://3ds.guide/

dothackjhe, 2017-03-18:
@Margen67 I tried booting Luma3DS via MSET using a new micro-SD Card, it still would not boot even if I used Luma3DS.nds with my DS Flash Cart and patch the system for 6.X Luma3DS.

dothackjhe, 2017-03-18:
[Update] I just successfully transitioned from rxTools to Luma3DS via A9LH route. Going through the process, I know better now which part of it risks anyone's handheld of bricking their device. 

This video tutorial saved me from that trouble: https://www.youtube.com/watch?v=DWq6bhSJ7Bo&t=1141s

Margen67, 2017-03-18:
I'm glad that worked for you but you should just use https://3ds.guide/ video tutorials are outdated and more likely to brick you.

closed 2017-03-18 by TuxSH