LumaTeam/Luma3DS #381: "Enable game patching" is gone from my options

Salamance73, 2017-03-15:
[image]

This is not my first time booting and I'm on o3ds.

Salamance73, 2017-03-15:
Should also mention I'm on 11.3

Margen67, 2017-03-15:
https://github.com/AuroraWright/Luma3DS/wiki/Options-and-usage

Salamance73, 2017-03-15:
So is it just gone or what?

pokecraft98, 2017-03-15:
You are using the stable version of luma. they do the same thing.

pokecraft98, 2017-03-15:
The 5th option is still game patching. Its just a different name.

Salamance73, 2017-03-15:
I have the titles folder and everything, but my game is not getting patch no matter what I select in the luma menu.

sora10pls, 2017-03-15:
You're using an older release of Luma3DS. The file structure you have on your SD Card matches those of the nightlies (`SD:/luma/titles/`), whereas the release you're showing off in the pic uses the directory from the latest stable release, v6.6 (`SD:/luma/code_sections`).

Update to latest nightly to fix, this is not an issue.

Salamance73, 2017-03-15:
Thank you, I had a feeling I wasn't using the right release when I updated. This also may have fixed my problem with homebrew menu accessibility.

closed 2017-03-15 by Salamance73