LumaTeam/Luma3DS #384: [HELP] Need older version of Luma3DS

dothackjhe, 2017-03-18:
So I'm having trouble booting my unit to Luma3DS via MSET using its latest build and was thinking that it was patched to be so. Where do I get Luma3DS 6.31 (or any particular version in general) when this repository no longer hosts it?

Shadow322, 2017-03-18:
Your better off installing soundhax to boot Luma3DS via hbl

Margen67, 2017-03-18:
https://3ds.guide/

khang06, 2017-03-18:
just boot it from homebrew launcher, even that's better than mset...

dothackjhe, 2017-03-18:
My unit is still on 8.1.0-19U. I've no plans of updating the sysnand firmware to 9.2 just to enable menuhax or soundhax with it. I've messed up one O3DS before so I'm quite wary with making significant changes with my existing units. This is one reason I'm trying to stick to MSET.

TuxSH, 2017-03-18:
Luma3DS should work with mset, you need both the .dat and arm9loaderhax.bin (in spite of the name, as the .dat will launch arm9loaderhax.bin)

closed 2017-03-18 by TuxSH

pokecraft98, 2017-03-18:
https://3ds.guide/decrypt9-(browser)# Do this for goodness sake

Margen67, 2017-03-18:
@pokecraft98 He's already installed A9LH. https://github.com/AuroraWright/Luma3DS/issues/380#issuecomment-287525156

dothackjhe, 2017-03-18:
@TuxSH Tried that many times, even with a new micro-SD Card, it did not work. I dunno what's causing it not to. Thanks to that persistent error, I was committed to go for Luma3DS via A9LH route. Luckily, the process went very smoothly with me.

TuxSH, 2017-03-18:
Weird.

dothackjhe, 2017-03-18:
@TuxSH True. The reason I even jumped to Luma3DS from rxTools via MSET route was because someone told me that the steps I needed for it only involved replacing files fitted for Luma3DS 6.6: sadly, it did not work with me.