LumaTeam/Luma3DS #388: Romfs not working on nightlies

Nyquis, 2017-03-21:
I have tested this on two nightlies, the 5081439f and 0eab9127 both do not seem to load even vanilla romfs. I extracted romfs with godmode9 then copied it to titleid/romfs.

Enable game patches is enabled.

Currently the game loads, gets past nintendo splashcreen and _loading data on sd_ only to proceed to a black screen with both blue yellow led turned on. My tomfs file is 1.7gb.

Game tested on. Fire Emblem Fates.

jpmac26, 2017-03-23:
I haven't experienced a 311 error, but I have been having issues getting external romfs patches working on the latest nightlies, so I can likely corroborate that part of the issue.

AuroraWright, 2017-04-14:
The buggy RomFS redir was replaced by LayeredFS

closed 2017-04-14 by AuroraWright