LumaTeam/Luma3DS #390: Issue Compiling.

HugaShire, 2017-04-02:
[image]

I have followed the instructions on the README.md

Shadow322, 2017-04-02:
If u didn't modify the code just use lumaupdate via 3ds

HugaShire, 2017-04-02:
I tried that, when ever I use Lumaupdater, out of region games that uses locals, homebrew and more break.

Shadow322, 2017-04-02:
The path has changed and I don't know what it is

urherenow, 2017-04-02:
The directory name "Luma3ds-master" leads me to believe that you downloaded the source .zip file instead of cloning. Did you begin with this?

git clone --recursive https://github.com/AuroraWright/Luma3DS

HugaShire, 2017-04-02:
@urherenow I did that.

HugaShire, 2017-04-02:
@urherenow I get fatal: unable to access 'https://github.com/AuroraWright/Luma3DS.git/': error setting certificate verify locations:
  CAfile: C:\Users\USER\AppData\Local\GitHub\PortableGit_f02737a78695063deace08e96d5042710d3e32db\mingw32/usr/ssl/certs/ca-bundle.crt
  CApath: none

TuxSH, 2017-04-02:
That's a git/system configuration problem, not a Luma one (referring to your last msg)

Otherwise just edit the Makefile and replace the part with -DCOMMIT_HASH to -DCOMMIT_HASH="0".

HugaShire, 2017-04-02:
$(dir_build)/patches.o: CFLAGS += -DREVISION=\"$(revision)\" -DCOMMIT_HASH="0x$(commit)" That's what it has

HugaShire, 2017-04-03:
<command-line>:0:13: error: invalid suffix "x" on integer constant
source/patches.c:329:24: note: in expansion of macro 'COMMIT_HASH'
     info->commitHash = COMMIT_HASH;
                        ^~~~~~~~~~~
make: *** [build/patches.o] Error 1

C:\Users\Cheryl\Desktop\Luma3DS-master>

TuxSH, 2017-04-11:
yeah replace "0x$(commit)" by "0" if you don't have a proper git install

closed 2017-04-11 by TuxSH