LumaTeam/Luma3DS #392: Can't boot without SD with Luma in CTRNAND

chronoss09, 2017-04-05:
I can't launch Luma without SD card, I have the file "arm9loaderhax.bin" from Luma in SYSNAND CTRNAND but it can't boot :(, i use the last build...

Margen67, 2017-04-05:
Are you sure you have the latest Luma nightly and newest arm9loaderhax version?

chronoss09, 2017-04-11:
For Luma yes but i don't know for arm9loaderhax, i will check it...

chronoss09, 2017-04-13:
Tested with latest Luma (28e6ad3) and no boot :(

Margen67, 2017-04-13:
Are you sure you're pasting arm9loaderhax.bin to the root of CTRNAND?

chronoss09, 2017-04-13:
Yes, pasting with GodMod9. I used Luma 6.6 and the last in commint....

Margen67, 2017-04-13:
Can you try using FBI to paste arm9loaderhax.bin?

chronoss09, 2017-04-13:
Samething......

Margen67, 2017-04-13:
Are you using a custom path?

chronoss09, 2017-04-14:
No, I just copy Luma in CTRNAND, nothing else...

Margen67, 2017-04-14:
Try deleting config.bin from the luma folder on the SD card.

chronoss09, 2017-04-14:
I want booting without SD card and i think config.bin hasn't a problem because he is on my SD card :(

luigoalma, 2017-04-15:
did you update a9lh?

chronoss09, 2017-04-15:
A9lh or Luma ?

and how to update a9lh ?

I test the same file on N3DS and it work fine, i don't know why it don't work on O3DS :(

luigoalma, 2017-04-15:
well, if you want to be certain about it
https://3ds.guide/updating-a9lh

chronoss09, 2017-04-16:
ok but i'm already up date because i have all these files...

MelonGx, 2017-04-16:
Both Data Input V3 and V4 support SD/MicroSD-less boot if Luma's arm9loaderhax.bin is in the root of CTRNAND.

OP should have not updated his A9LH to Data Input V3 or V4.

closed 2017-04-17 by AuroraWright