LumaTeam/Luma3DS #397: MenuHax3.2 and Luma3DS 11.3 wont boot in the HomebrewLauncher

bliblablub231, 2017-04-12:
Cant boot into the Homebrew Launcher since the update. I only can configure the MenuHax options by pressing select. All the other buttons doesnt seem to work. Any solutions? The only thing i changed during the last session was the Splash.bin because the i wanted to try out my custom made splash screen. Does anyone know how to fix this?

Posted it in the Subreddit before, but my Thread got deleted for some reason. Just trying to get some help.
Proof: https://www.reddit.com/r/3dshacks/comments/64yji0/changed_my_3ds_theme_and_now_homebrew_wont_open/

Margen67, 2017-04-12:
1. Stop using menuhax
2. https://3ds.guide/

MelonGx, 2017-04-16:
3. Don't mind what Margen67 said.
4. boot.3dsx bundled by 7.0 release is crashed. Use old version's instead.

AuroraWright, 2017-04-17:
Fixed

closed 2017-04-17 by AuroraWright