LumaTeam/Luma3DS #398: Game patching doesn't work on hourlies

McFork, 2017-04-14:
I've tried 00dbbe1b and 5b6bd04 because my N3DS wouldn't launch NDS games anymore, and language emulation is broken on both those versions.
I have game patching enabled, games run well (except for Majora's mask that needs the emulation to start), but are always in the system languages despite the locales being changed.
Also, the Umbra Moon mod is now deactivated on Pokémon Moon, which would mean that the whole game patching functionnality is broken.

Shadow322, 2017-04-14:
It's not broken just the path has changed and I don't remember the path details

McFork, 2017-04-14:
Ah really? The location in the wiki is still the same, do you know where we can find the new one?
Thanks for your answer

urherenow, 2017-04-14:
How about in the 20 other closed issues that have been posted and answered about this very thing already?

Look at the closed issues.

Margen67, 2017-04-14:
Also: ``
 Please make sure to read "Enable region/language emulation and external .code" https://github.com/AuroraWright/Luma3DS/wiki/Options-and-usage before posting any issues about the "Enable region/language emulation and external .code" option(s).
 Keep in mind that Wiki page only applies to nightly builds. It will NOT WORK with 6.6 Stable Luma3DS.
 As of 0.02 (https://github.com/Possum/LumaLocaleSwitcher/releases) LumaLocaleSwitcher's path(s) are for 6.6 Stable and won't work with Luma3DS nightly builds newer or equal to https://github.com/AuroraWright/Luma3DS/commit/b5336c81cc82b6c5e8115249342beb5b065cdce9.
 Use this version for Luma3DS nightlies newer or equal to https://github.com/AuroraWright/Luma3DS/commit/b5336c81cc82b6c5e8115249342beb5b065cdce9 :
 https://puu.sh/uC5zW/5470adc347.7z (from https://github.com/Possum/LumaLocaleSwitcher/issues/9#issuecomment-285564014)
``

Possum, 2017-04-14:
@Margen67 
> Keep in mind that Wiki page only applies to nightly builds. It will NOT WORK with 6.6 Stable Luma3DS. 

I think you have that backwards - that wiki page seems to have the old (6.6 "stable") paths.

The new paths are "/luma/titles/0000000000000000/locale.txt", replacing the 0s with the actual title ID (see also #382).

Margen67, 2017-04-14:
@Possum The issue template needs to be updated.

suzsuzuki, 2017-04-14:
I can confirm region changing stop working.

I tested with nightly f8984525 and worked. With latest nightly f7cc2d29 didn't works! I didn't change directory structure /luma/titles/TITLE_ID/locale.txt, just update with luma updater and after that region change stop working

McFork, 2017-04-14:
Thanks for the reply @Margen67, I missed the template indeed, sorry about that.
However, unless there is something else I missed, there still seems to be a problem. I placed the locales file at the right location (/luma/titles/TITLE_ID/locale.txt) with the right values inside ("EUR FR", separated with a space, no carriage return), I double checked that game patching is enabled, and still the new language isn't emulated.
I took screencaps of the file's path and content that I can share with you: [http://imgur.com/sy2Q0od](http://imgur.com/sy2Q0od)

suzsuzuki, 2017-04-14:
I can confirm it working know. AuroraWright fixed the issue on Luma:

https://gbatemp.net/threads/luma3ds-noob-proof-3ds-custom-firmware.411110/page-921#post-7241741

closed 2017-04-14 by AuroraWright