LumaTeam/Luma3DS #400: Infinite Loading on LayeredFS

Tech0verlord, 2017-04-15:
So seeing as LayeredFS was about to be officially released, I went ahead and downloaded the latest nightly (at the time of writing ad1dc51e) and proceeded to try out the new feature. Seems like it might have an issue with out of region games, or just the two games in question. 

I can confirm I LayeredFS is working as with Pokémon X (US region), I can get the neo X mod running. However, when I try to load the English Patch for Monster Hunter Stories and Monster Hunter Double Cross (both Japanese-region games) , my 3ds gets stuck loading infinitely at the 3ds logo. I'm not sure what could be the issue, as both patches work via NTR. I copied the appropriate files to the correct locations (as if I were replacing files in a romfs for a modded cia or hans redirection), so luma/titles/titleID/romfs/fileStructure. 

Info about my 3ds:
Model: New XL
FW: 11.3
Region: US
Luma version: 6.6-ad1dc51e

Tech0verlord, 2017-04-15:
I would like to note that when I originally used the test build Aurora put up on GBAtemp, for both monster hunter games, immediately after the 3ds logo I get that exception handler screen that she was looking for.

AuroraWright, 2017-04-15:
@TuxSH took a look at it, and it appears those games have a needed function (where the LayeredFS code is copied to) compiled to Thumb instead of regular ARM; the Thumb function could probably be found like the code now does for the ARM one, but  Thumb code is smaller than ARM and so this function isn't even capable of holding the LayeredFS code. So I'll leave this issue open just in case but I don't think there's anything that can be done

svanheulen, 2017-04-15:
So it looks like LayeredFS works by replacing the `throwFatalError` function in the game. The issue being that it's searching for the ARM version of the function and not finding it. And even if it was modified to search for the Thumb version the replacement would need to be Thumb as well (otherwise the game would crash when trying to call it).

There would need to be an ARM and a Thumb mode version of the payload and search functions for each. All the FS functions would most likely be in the same ARM/Thumb mode since they're all probably part of the same library in the SDK. But is the `throwFatalError` function part of a different library? If they are it might be a different mode than the rest of the functions which would be a pain to deal with.

AuroraWright, 2017-04-15:
For these games it finds all the symbols except for `throwFatalError`. Tux found it but it's thumb, and the ARM payload is too small to fit. I don't know if a thumb payload could be made in enough space to fit. That function is used because it's basically unused iirc, so we can use it for storage.

svanheulen, 2017-04-15:
Ah ok, I see it now, it's `fsOpenFileDirectly` and `fsTryOpenFile` that are patched. And the game would only call `throwFatalError` when things a irrecoverably messed up anyway.

So it would need a different function to replace or a smaller payload.

Tech0verlord, 2017-04-15:
If I may interject, what does the NTR LayeredFS plugin do different for romfs file redirection that luma doesn't seem able to do at the moment?

svanheulen, 2017-04-16:
One option would be using the padding at the end of the .text segment. Not every game will have enough space there but it's easy to check and it could just be a fallback.

svanheulen, 2017-04-16:
My patch should solve the issue for most games but it's always possible that a game would have the Thumb version of `throwFatalError` and not have enough padding.

Tech0verlord, 2017-04-17:
If this pull is available on the current hourly, I'll gladly test it on MHXX and MH Stories and report back. It'll be some time, a bit busy with stuff atm

Tech0verlord, 2017-04-17:
Well, tested using hourly builds 0dc07830 and fdbe4342 and I'm still hanging for MHXX.

svanheulen, 2017-04-17:
@Tech0verlord Do you have the update installed for MHXX? It's working fine for me in 0dc0783 but only with the update installed. I think it's an issued with the files, not Luma. Monster Hunter Stories is also working fine for me with the English patch.

Tech0verlord, 2017-04-17:
Yeah, I have the update installed, as that's what the English patch is now built against. Let me try with that new build. Does the game start up immediately, or does it take some time for you once the 3ds logo starts up?

svanheulen, 2017-04-17:
Yeah it starts up just as fast as when I have game patching turned off. I'm using completely unmodified versions of the game and update installed from the CDN.

Tech0verlord, 2017-04-17:
Well, it seems like this patch works. Hopefully this was an isolated case with the Monster Hunter games and no other games were affected. I guess I'll close this issue for now. Thanks for all of your hard work all.

closed 2017-04-17 by Tech0verlord